mavlink/MAVSDK #322: DroneCore Application template

shakthi-prashanth-m, 2018-03-15:
We don't have a standard layout for applications yet. If you have a look at any Android application for eg, it has a layout something like, 
```java
public class BasicClickHandlersActivity extends Activity {
  
	@Override
	protected void onCreate(Bundle savedInstanceState) {
		super.onCreate(savedInstanceState);
		setContentView(R.layout.activity_basic_click_handlers);
		Button secondButton = (Button) findViewById(R.id.btnClick2);
		secondButton.setOnClickListener(new View.OnClickListener() {
			@Override
			public void onClick(View v) {
				secondButtonClicked(v);
			}
		});
	}

	@Override
	public boolean onCreateOptionsMenu(Menu menu) {
		// Inflate the menu; this adds items to the action bar if it is present.
		getMenuInflater().inflate(R.menu.activity_basic_click_handlers, menu);
		return true;
	}
}
```
Its more clear for reader to visualize application's activity. Its clear that `onCreate()` is the entry point and application's business logic goes into one these callbacks. 

All DroneCore applications primarily deal with discovered device.

**Note:** I use term device until we merge code containing `System` terminology.

So, application won't do anything without device discovery. If there's no device discovered for 3 seconds, DroneCore fires timeout callback. So, its convenient if we layout a DroneCore application template similar to a below one:

[image]

And, names `on_discover()`, `on_timeout()` seems more natural than `register_on_discover()`, etc.
Kindly share your thoughts on it.
Thanks.

hamishwillee, 2018-04-03:
We support this paradigm obviously, but there is a reason why promises (as used in our examples) have become a popular tool for managing async behaviour. I think our examples are more clear than this - certainly more clear than this would be if expanded into a full application. 

> And, names on_discover(), on_timeout() seems more natural than register_on_discover(), etc.

I slightly agree.

julianoes, 2018-08-17:
Closing this for now. We'll have example applications such as the Swift example which show how to use the SDK.

closed 2018-08-17 by julianoes