mavlink/MAVSDK #325: Re-organize System/Device class interface for plugins and applications

shakthi-prashanth-m, 2018-03-16:
Header `system.h` gets installed along with `dronecore.h` and plugin header files.
DroneCore apps do:
```cpp
System &system = dc.system();
...
// have access to entire System interface
```
This implies, we're letting applications access to `System` interface as well. In other words, _applications can even compose MAVLink messages and send it across!_
I think thats not right. We may decide to export minimal interface like, `has_camera(1)`, `has_gimbal()`, etc;

As I understand, we insist DroneCore apps to use plugins rather than `System` interface. Please correct me.

On the other hand, `System` interface is needed for plugins. So, one possible way is to expose:
```cpp
class MAVLinkSystem {
// Current Device contents which are required by plugins goes here.
};
```
to plugins.
**Note:** _I chose name `MAVLinkSystem` here because its internal to DroneCore and not exposed to public. I am fine with any suitable name that makes sense._

For DroneCore applications, lets provide 
```cpp
class System {
// Minimal interface needed for DroneCore apps
};
```
Kindly share your thoughts.

hamishwillee, 2018-03-19:
I tend to agree with you. I don't mind that app authors can use it, I mind that if they do so an this is a "public interface" we're gong to effectively be making compatibility promises about that interface going forward.

shakthi-prashanth-m, 2018-03-19:
@julianoes @JonasVautherin @hamishwillee 
I've got these changes (on top of #310) right now (Tested). I shall submit.

shakthi-prashanth-m, 2018-03-19:
Current design: (Details omitted)
[image]

Proposed design: (Details omitted)
[image]

commit 2e54aa70 (2018-03-19): core, plugins: Re-organize System class to fix #325

This makes System class to expose generic public interface pushing
back MAVLink details as part of `MAVLinkSystem` class. An application
which gets an instance of `System` class will only access to only
required minimal interface. Plugins now use `MAVLinkSystem` class
for interacting with PX4 Autopilot.
commit f8b78318 (2018-03-23): core, plugins: Re-organize System class to fix #325

This makes System class to expose generic public interface pushing
back MAVLink details as part of `MAVLinkSystem` class. An application
which gets an instance of `System` class will only access to only
required minimal interface. Plugins now use `MAVLinkSystem` class
for interacting with PX4 Autopilot.
commit aa91cb21 (2018-04-03): core, plugins: Re-organize System class to fix #325

This makes System class to expose generic public interface pushing
back MAVLink details as part of `MAVLinkSystem` class. An application
which gets an instance of `System` class will only access to only
required minimal interface. Plugins now use `MAVLinkSystem` class
for interacting with PX4 Autopilot.
commit f43921cc (2018-04-03): core, plugins: Re-organize System class to fix #325

This makes System class to expose generic public interface pushing
back MAVLink details as part of `MAVLinkSystem` class. An application
which gets an instance of `System` class will only access to only
required minimal interface. Plugins now use `MAVLinkSystem` class
for interacting with PX4 Autopilot.