mavlink/MAVSDK #329: Rename "Mavlink" stuffs to that of "MAVLink"

shakthi-prashanth-m, 2018-03-20:
I see names of class/method/enum, etc contain **MAVLink*** in QGroundControl source code.
Its there even in app console logs too.
```
[11:40:51|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m (device.cpp:247)
[11:40:51|Debug] MAVLink: info: Takeoff detected (device.cpp:247)
[11:40:51|Debug] MAVLink: critical: Using minimum takeoff altitude: 2.50 m (device.cpp:247)
[11:40:51|Debug] MAVLink: info: data link #1 lost (device.cpp:247)
```
I named `class MAVLinkSystem` for the same reason in  #310.
We've these to rename:
```
class MavlinkCommands {...};
class MavlinkParameters {...};
class MavlinkReceiver {...};
class MavlinkChannels {...};
```

Please suggest: @hamishwillee @julianoes @JonasVautherin .
Thanks.

hamishwillee, 2018-03-20:
If MAVLink is always implied then it isn't needed. If something is intended to be generic then it shouldn't be prefixed. You'd use a MAVLink where something must always be for working with MAVLink and only makes sense in that context.

shakthi-prashanth-m, 2018-03-20:
OK. Thanks for rephrasing my question @hamishwillee . :)

shakthi-prashanth-m, 2018-03-20:
Addressed as part of #310.

julianoes, 2018-03-22:
To me mavlink is sort of like a word which is why I wrote it that way. We can change it but I don't know if the churn is really warranted.

hamishwillee, 2018-03-22:
Probably not, but already done :-)

shakthi-prashanth-m, 2018-04-05:
Addressed in #310 .

closed 2018-04-05 by shakthi-prashanth-m