mavlink/MAVSDK #331: Maintain a connection instance per component

shakthi-prashanth-m, 2018-03-22:
Right now we've maintain one connection per System (one or more components).
This creates problem if there are more than one component in a system.

Consider below scenario:

- After we receive heartbeat from a component A of a system on udp port (say, 14540), we fill source ip/port details as part of connection instance. When we receive heartbeat from another component B of the same system, **_Core assumes System IP may be changed due to Wifi disconnection._**
- While we're sending commands/messages, we need to make sure we send to IP/Port of the corresponding component whom it is meant to be.

I think, to support multi-component in core, we shall have one connection per component. We should map each component with its connection to make this happen.

Please provide better suggestions, if any.
It is linked to #304 . And hopefully the fix should be part of #310 .

shakthi-prashanth-m, 2018-03-22:
How about maintaining Remote client info as part of class `UdpConnection` ?
```cpp
class UdpConnection : Connection  {
    ....
    struct RemoteClient {
        uint8_t sysid = 0;
        uint8_t compid = 0;
        std::string ip = {};
        int port = 0;
    };
    std::vector<RemoteClient> _remote_clients = {};
};
```
instead of:
```cpp
class UdpConnection : Connection  {
    ....
    std::string _remote_ip = {};
    int _remote_port_number = 0;
};

```
We then would send messages only on that remote ip/port of that component.

julianoes, 2018-03-22:
Thanks @shakthi-prashanth-m. This is indeed something to think about. We're getting into the mavlink networking territory which is anything but trivial and not properly specced.

We can also have cases where there are two connections, one UDP (wifi) and one serial (a long range radio) to the same system possibly including more than one component.

The other question is if we should ever send commands to a specific component or if we can always send it on all connections and hope it will arrive in some way.

shakthi-prashanth-m, 2018-03-22:
>We can also have cases where there are two connections, one UDP (wifi) and one serial (a long range radio) to the same system possibly including more than one component.

You mean, application would appx will do:
```
dc.add_udp_connection(14540);
dc.add_serial_connection("/dev/ttyS1", 921600);
```
I think this would be handled by #310.

I saw the below problem while dealing with camera component.
```
Note: Google Test filter = Camera.ReceiveCameraInformation
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from Camera
[ RUN      ] Camera.ReceiveCameraInformation

[05:46:07|Info ] New system on: 127.0.0.1:14556 (udp_connection.cpp:206)
[05:46:07|Debug] New: System ID: 1 Comp ID: 1 (dronecore_impl.cpp:312)
[05:46:07|Debug] Component Autopilot added. (mavlink_system.cpp:318)
[05:46:07|Error] Remote IP unknown (udp_connection.cpp:135)

[05:46:07|Error] send fail (dronecore_impl.cpp:94)
[05:46:07|Info ] System changed to: 127.0.0.1:37043 (udp_connection.cpp:217)

[05:46:07|Debug] Component Camera 1 added. (mavlink_system.cpp:318)
[05:46:07|Info ] System changed to: 127.0.0.1:14556 (udp_connection.cpp:217)

[05:46:08|Error] Remote IP unknown (udp_connection.cpp:135)
[05:46:08|Error] send fail (dronecore_impl.cpp:94)
[05:46:08|Error] connection send error (520) (mavlink_commands.cpp:262)
[05:46:08|Debug] # of components: 2 (mavlink_system.cpp:427)
[05:46:08|Debug] Discovered 4294967298 (dronecore_impl.cpp:333)
[05:46:08|Error] Remote IP unknown (udp_connection.cpp:135)
[05:46:08|Error] send fail (dronecore_impl.cpp:94)
[05:46:08|Info ] System changed to: 127.0.0.1:37043 (udp_connection.cpp:217)
[05:46:08|Info ] System changed to: 127.0.0.1:14556 (udp_connection.cpp:217)
[05:46:09|Error] Remote IP unknown (udp_connection.cpp:135)
[05:46:09|Error] send fail (dronecore_impl.cpp:94)
[05:46:09|Info ] System changed to: 127.0.0.1:37043 (udp_connection.cpp:217)
[05:46:09|Info ] System changed to: 127.0.0.1:14556 (udp_connection.cpp:217)
[05:46:10|Error] Remote IP unknown (udp_connection.cpp:135)
[05:46:10|Error] send fail (dronecore_impl.cpp:94)
[05:46:10|Info ] System changed to: 127.0.0.1:37043 (udp_connection.cpp:217)
[05:46:10|Info ] System changed to: 127.0.0.1:14556 (udp_connection.cpp:217)
[05:46:11|Error] Remote IP unknown (udp_connection.cpp:135)
[05:46:11|Error] send fail (dronecore_impl.cpp:94)
[05:46:11|Info ] System changed to: 127.0.0.1:37043 (udp_connection.cpp:217)
[05:46:11|Info ] System changed to: 127.0.0.1:14556 (udp_connection.cpp:217)
[05:46:12|Error] Remote IP unknown (udp_connection.cpp:135)
[05:46:12|Error] send fail (dronecore_impl.cpp:94)
[05:46:12|Error] connection send error (521) (mavlink_commands.cpp:262)
Waiting simply...
[05:46:12|Warn ] command failed (521). (mavlink_commands.cpp:151)
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: Promise already satisfied
Aborted (core dumped)
```

In the above logs:
**Remote IP:**  127.0.0.1 (localhost)
**PX4 Autopilot Remote port:** 14556
**Camera Remote port:** 37043

As we maintain only instance per connection, we keep overwriting onto it when received data from different components (same system or different).

shakthi-prashanth-m, 2018-03-23:
I modified the issue to consider UDP connection only, because in case of UDP connection, we have chances of multiple components sending to a common port as below:

[image]

shakthi-prashanth-m, 2018-03-28:
As discussed in WG Meeting, we're leaving this feature to [MAVLink Router ](https://github.com/intel/mavlink-router) as it is meant to handle this feature. It depends on how many manufactures on companion board adopt MAVLink Router. Ideally, its the job of MAVLink Router, not DroneCore.

In absence of MAVLink Router these changes would be required. For now, we're closing this issue to keep DroneCore to have less complications. We may add it in future if required.

commit ab2c4834 (2018-03-23): core, plugns, integration_tests: Fix #331

Now DroneCore handles multiple component requests on a same UDP port.
UDPConnection was only holding one instance of remote client, but
now we create so many instances of remote client as many clients
send requests. Also, when replying back, now we reply only to the
intended component.
closed 2018-03-28 by shakthi-prashanth-m