mavlink/MAVSDK #339: DroneCore will not build on new Raspberry Pi 3 B+

conorf54, 2018-03-25:
I have used DroneCore on Raspberry Pi 3 B, but on the new Raspberry Pi 3 B+ DroneCore will not install and keeps rebooting when trying to run the line below. 

git submodule update --init --recursive

Does anybody know what is causing this issue?

conorf54, 2018-03-25:
> No, but that line is simply running the git tool - effectively nothing to do with DroneCore at all. You need to  ask for git support/search google for issues related to git and Ra Pi.

Git works with every application except DroneCore, but I will delete it and re download see if that fixes the issue 
		
		

		Conor FergusonCEO/President, WISPr SystemsPhone: 662-654-5702

hamishwillee, 2018-03-25:
I guess another possibility is that you're running out of memory or similar.

conorf54, 2018-03-25:
Well running it on the previous pi 3 B DroneCore ran perfect on it, the new pi 3 B+ I am having issues still, but I will update after a solution has been found 
		
		

		Conor FergusonCEO/President, WISPr SystemsPhone: 662-654-5702

JonasVautherin, 2018-05-25:
What's the state of this? Did you try it again recently?

conorf54, 2018-05-25:
Yes, so the problem I was having was due to the power supply I was using.  The new RPi 3B+ must use at least 1.5A to 2A and 5V power supply to power the RPi 3B+ or it doesn’t have enough power to compile.

conorf54, 2018-05-25:
So 5V/2A power supply solved the problem

julianoes, 2018-05-25:
Great, thanks for the feedback @conorf54!

closed 2018-05-25 by julianoes