mavlink/MAVSDK #345: Create Appropriate DNS URL/redirects for Dronecode SDK

hamishwillee, 2018-04-04:
DroneCore has two entry points:
1. http://dronecore.io/ (website)
2. https://docs.dronecore.io (docs)

The website should be found at dronecode.org/sdk
- Is the plan to move this inside the dronecode site (abandon the old site?). In this case the actions are:
    - merge the dronecore.io page into dronecode site
    - create a redirect from dronecore.io to the new page on dronecode site
@mrpollo can you confirm the above? If that is correct I will start creating a new page in dronecode.org/sdk. You would sort out the redirect.

For the docs, I presume we just need new URL and redirect of old URL to that new URL. Not sure what was agreed - sdkdocs.dronecode.org? docs.dronecode.org? docs.dronecode.org/sdk?

hamishwillee, 2018-04-13:
FYI, according to @mrpollo 
- the dronecore page should be updated to "sdk" and embedded into the dronecode site at /sdk
- docs are to be moved to docs.droncode.org (This may also include the camera manager docs).

mrpollo, 2018-04-13:
Agreed, this *includes* the camera docs

mrpollo, 2018-04-13:
Who owns dronecore.io? @LorenzMeier was that you?

mrpollo, 2018-04-13:
Docs URL: docs.dronecode.org
Marketing SDK: dronecode.org/sdk
Marketing Camera: dronecode.org/camera-manager (or whatever name it gets)

julianoes, 2018-05-30:
What is the status here?

hamishwillee, 2018-05-31:
Currently @mrpollo is waiting on Linux Foundation to create the new URLs for camera manager and dronecode sdk. He has pinged them several times. 

Originally we were after docs.dronecode.org for combined camera manager and SDK docs. However that won't work because dronecore has versioning, and we can't support different versioning within the one docs package. I can't remember exactly the what the final urls Ramon asked for, but I _think_ it was sdk.dronecode.org and camera-manager.dronecode.org.

@mrpollo Can you go bollock LF again? Can we escalate - this is silly.

mrpollo, 2018-05-31:
Yes will do. Thanks for the update team.

mrpollo, 2018-07-09:
Only thing left to do is to redirect dronecore.io to dronecode.org/sdk I don't own the dronecore.io domain I think that got purchased by @LorenzMeier

julianoes, 2018-07-24:
@LorenzMeier has done the redirects, closing.

closed 2018-07-24 by julianoes