mavlink/MAVSDK #347: Improving DroneCore logging

shakthi-prashanth-m, 2018-04-04:
Right now, we're logging DroneCore logs on console. Generally its a practice to dump all the debug logs to a file. We could save them onto say, `build/default/logs/dronecore.log`.

Then, a user may simply run on a terminal:
```tail -f build/default/logs/dronecore.log```

So, we can add **more debug/info logs** in the code (as it won't collide with application console logs) which are really helpful to analyze issues. For ex: say, arming/takeoff fails. We can easy look into logs to see whether **all the necessary things went well** or not.

PS: _As discussed in Slack, we're planning to add this in future (not right now)._

julianoes, 2019-04-24:
Additionally, we should add a method to specify the log level.

julianoes, 2024-05-27:
Implemented, see example https://github.com/mavlink/MAVSDK/tree/main/examples/log_callback.

closed 2024-05-27 by julianoes