mavlink/MAVSDK #353: Problems connecting to multiple PX4s in SITL

varunvp, 2018-04-07:
I want to write a DroneCore program to control a swarm of SITL PX4s simultaneously, through offboard mode. Right now, I'm using a swarm of 4 drones, which is a modification of the scripts here:https://dev.px4.io/en/simulation/multi-vehicle-simulation.html. I've been using the takeoff_land example program from DroneCore as a base, When I run the example program, the first drone with a sysid of 1 takes off, which is as expected. Now if I have to control all the 4 drones, I have to add the 4 UDP ports of the drones, which I did, as below.
```
DroneCore::ConnectionResult connection_result = dc.add_udp_connection(14540);
    dc.add_udp_connection(14541);
    dc.add_udp_connection(14542);
    dc.add_udp_connection(14543);
```

Of all these, I just want to add device for one drone, for the sake of testing. 
`    Device &device = dc.device(1/2/3/4(mavlink sys ids));`
Now here comes the problem. It seems that all the PX4 SITL drones have the same UUID, so even differentiating between the MAVLink sys id 's is useless, because DroneCore won't even consider them. So I had to comment out this part of code from device_impl.cpp, so that it forces use of sys id.
```
// if (_target_uuid == 0 && autopilot_version.uid != 0) {

    //     // This is the best case. The device has a UUID and we were able to get it.
    //     _target_uuid = autopilot_version.uid;

    // }
```
Even after doing all this, I start getting the following stream of errors, when I use any sys id other than 1 to connect.
This is the error beginning

```
Waiting to discover device...
[06:21:42|Info ] New device on: 127.0.0.1:14553 (udp_connection.cpp:211)
[06:21:42|Error] Remote IP unknown (udp_connection.cpp:140)
[06:21:42|Error] send fail (dronecore_impl.cpp:104)
[06:21:42|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Info ] New device on: 127.0.0.1:14555 (udp_connection.cpp:211)
[06:21:42|Error] Remote IP unknown (udp_connection.cpp:140)
[06:21:42|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[06:21:42|Error] send fail (dronecore_impl.cpp:104)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:211)
[06:21:42|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[06:21:42|Error] Remote IP unknown (udp_connection.cpp:140)
[06:21:42|Error] send fail (dronecore_impl.cpp:104)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Info ] New device on: 127.0.0.1:14559 (udp_connection.cpp:211)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 lost (device_impl.cpp:246)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 lost (device_impl.cpp:246)
[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 lost (device_impl.cpp:246)
[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 lost (device_impl.cpp:246)
[06:21:42|Debug] MAVLink: info: data link #0 regained (device_impl.cpp:246)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 regained (device_impl.cpp:246)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 regained (device_impl.cpp:246)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:21:42|Debug] MAVLink: info: data link #0 regained (device_impl.cpp:246)
[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)

```
This is the error end
```
[06:20:22|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:20:22|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:20:22|Error] Retrying failed (511) (mavlink_commands.cpp:215)
[06:20:22|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:20:22|Error] Error: set param busy timeout: NAV_FT_RS (mavlink_parameters.cpp:413)
[06:20:22|Error] Failed to set NAV_FT_RS: 0.5 (follow_me_impl.cpp:266)
Setting rate failed:Timeout
```
DroneCore is able to detect the presence of other devices, but something amiss happens. What could the problem be? I've uploaded the cpp program for reference.
[Keyboard_test.txt](https://github.com/dronecore/DroneCore/files/1886374/Keyboard_test.txt)
Thanks!

shakthi-prashanth-m, 2018-04-09:
@varunvp Are you using latest develop or master ?

> so even differentiating between the MAVLink sys id 's is useless, because DroneCore won't even consider them. 

Why ? If UUID is not available, then DroneCore uses `sysid` to differentiate vehicles. I'm not sure, but can you disable providing UUID?

>[06:21:42|Debug] sysid: 2 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 3 (dronecore_impl.cpp:90)
[06:21:42|Debug] sysid: 4 (dronecore_impl.cpp:90)

It is receiving heartbeats from 4 vehicles, hence it is printing their sysid's. (It doesn't print the first one).

>[06:21:42|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[06:20:22|Error] Error: set param busy timeout: NAV_FT_RS (mavlink_parameters.cpp:413)
[06:20:22|Error] Failed to set NAV_FT_RS: 0.5 (follow_me_impl.cpp:266)

Are you using Follow Me plugin ? It is failing to set Follow Me param. Can you take fresh master and try ? OR You can give it a try on latest develop as well.

varunvp, 2018-04-09:
At the time of writing, I was using the latest master branch. 

> Why ? If UUID is not available, then DroneCore uses sysid to differentiate vehicles. I'm not sure, but can you disable providing UUID?

I am not aware of any methods of disabling UUID. We went searching for it in the PX4 source, and found it to be '#define'd as a constant. Didn't want to mess with it.

> Are you using Follow Me plugin ? It is failing to set Follow Me param. Can you take fresh master and try ? OR You can give it a try on latest develop as well.

It automatically uses the FollowMe plugin, I don't know why.

I tried taking the fresh master after I had modified that file device_impl.cpp, but even after replacing the file with the original file, and building it, it seemed as if was still using the same file with the comment active. That's why I've downloaded it fresh, and I've got to try it now.

varunvp, 2018-04-09:
This is what I get when I switch to develop from master(just for your information)
```
varun@varun-5577:~/DroneCore/DroneCore$ git checkout develop
warning: unable to rmdir curl-android-ios: Directory not empty
warning: unable to rmdir libs/gtest: Directory not empty
warning: unable to rmdir libs/include/mavlink/v2.0: Directory not empty
warning: unable to rmdir libs/tinyxml2: Directory not empty
Switched to branch 'develop'
Your branch is up-to-date with 'origin/develop'.
varun@varun-5577:~/DroneCore/DroneCore$ git submodule update --init --recursive
Submodule 'dronecore-proto' (https://github.com/dronecore/DroneCore-Proto.git) registered for path 'backend/proto'
Submodule 'tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'core/third_party/tinyxml2'
Submodule 'json11' (https://github.com/dropbox/json11) registered for path 'plugins/mission/third_party/json11'
Submodule 'gtest' (https://github.com/google/googletest) registered for path 'third_party/gtest'
Submodule 'mavlink_v2' (https://github.com/mavlink/c_library_v2.git) registered for path 'third_party/mavlink/include/mavlink/v2.0'
Submodule 'zlib' (https://github.com/madler/zlib) registered for path 'third_party/zlib'
Cloning into 'backend/proto'...
remote: Counting objects: 106, done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 106 (delta 29), reused 45 (delta 24), pack-reused 54
Receiving objects: 100% (106/106), 13.66 KiB | 0 bytes/s, done.
Resolving deltas: 100% (31/31), done.
Checking connectivity... done.
Submodule path 'backend/proto': checked out '57ecd082bfc94d4b6cef7c58448091782b8e5d32'
fatal: reference is not a tree: 9dfcd9073f84bdb53064b2318db1f26801c67826
Cloning into 'core/third_party/tinyxml2'...
remote: Counting objects: 3500, done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 3500 (delta 27), reused 43 (delta 25), pack-reused 3446
Receiving objects: 100% (3500/3500), 2.65 MiB | 139.00 KiB/s, done.
Resolving deltas: 100% (2260/2260), done.
Checking connectivity... done.
Submodule path 'core/third_party/tinyxml2': checked out '37bc3aca429f0164adf68c23444540b4a24b5778'
Cloning into 'plugins/mission/third_party/json11'...
remote: Counting objects: 296, done.
remote: Total 296 (delta 0), reused 0 (delta 0), pack-reused 296
Receiving objects: 100% (296/296), 82.31 KiB | 33.00 KiB/s, done.
Resolving deltas: 100% (163/163), done.
Checking connectivity... done.
Submodule path 'plugins/mission/third_party/json11': checked out 'ec4e45219af1d7cde3d58b49ed762376fccf1ace'
Cloning into 'third_party/gtest'...
remote: Counting objects: 11103, done.
remote: Compressing objects: 100% (70/70), done.
remote: Total 11103 (delta 110), reused 163 (delta 108), pack-reused 10912
Receiving objects: 100% (11103/11103), 3.48 MiB | 183.00 KiB/s, done.
Resolving deltas: 100% (8149/8149), done.
Checking connectivity... done.
Submodule path 'third_party/gtest': checked out '0fe96607d85cf3a25ac40da369db62bbee2939a5'
Cloning into 'third_party/mavlink/include/mavlink/v2.0'...
remote: Counting objects: 12233, done.
remote: Compressing objects: 100% (99/99), done.
remote: Total 12233 (delta 288), reused 276 (delta 245), pack-reused 11889
Receiving objects: 100% (12233/12233), 5.66 MiB | 326.00 KiB/s, done.
Resolving deltas: 100% (11130/11130), done.
Checking connectivity... done.
Submodule path 'third_party/mavlink/include/mavlink/v2.0': checked out '0773d299bd2ebfb59a496803606783b04372774c'
Cloning into 'third_party/zlib'...
remote: Counting objects: 5045, done.
remote: Total 5045 (delta 0), reused 0 (delta 0), pack-reused 5045
Receiving objects: 100% (5045/5045), 2.52 MiB | 59.00 KiB/s, done.
Resolving deltas: 100% (3574/3574), done.
Checking connectivity... done.
Submodule path 'third_party/zlib': checked out 'cacf7f1d4e3d44d871b605da3b647f07d718623f'
Unable to checkout '9dfcd9073f84bdb53064b2318db1f26801c67826' in submodule path 'core/third_party/curl-android-ios'
```

When I execute with 
```
DroneCore::ConnectionResult connection_result = dc.add_udp_connection(14540);
...
Device &device = dc.device(2);
```

I get this
```
Waiting to discover device...
[01:07:41|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:211)
[01:07:41|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[01:07:41|Debug] MAVLink: info: data link #0 lost (device_impl.cpp:244)
[01:07:41|Debug] MAVLink: info: data link #0 regained (device_impl.cpp:244)
[01:07:41|Debug] Discovered 4294967298 (dronecore_impl.cpp:234)
Discovered device with UUID: 4294967298
[01:07:43|Error] device with UUID: 2 not found (dronecore_impl.cpp:178)
[01:07:43|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:178)
[01:07:43|Error] Error: more than one device found: (dronecore_impl.cpp:145)
[01:07:43|Warn ] strange: 0: 0, 0x2458c00 (dronecore_impl.cpp:149)
[01:07:43|Warn ] strange: 1: 1, 0x7fbbb0000e40 (dronecore_impl.cpp:149)
[01:07:43|Error] Error: set param busy timeout: NAV_MIN_FT_HT (mavlink_parameters.cpp:413)
[01:07:43|Error] Failed to set NAV_MIN_FT_HT: 8m (follow_me_impl.cpp:227)
[01:07:43|Info ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:197)
[01:07:44|Info ] sending again, retries to do: 2  (511). (mavlink_commands.cpp:197)
[01:07:44|Error] Error: set param busy timeout: NAV_FT_DST (mavlink_parameters.cpp:413)
[01:07:44|Error] Failed to set NAV_FT_DST: 8m (follow_me_impl.cpp:236)
[01:07:44|Error] Error: set param busy timeout: NAV_FT_FS (mavlink_parameters.cpp:413)
[01:07:44|Error] Failed to set NAV_FT_FS: Behind (follow_me_impl.cpp:257)
[01:07:44|Info ] sending again, retries to do: 1  (511). (mavlink_commands.cpp:197)
[01:07:45|Error] Retrying failed (511) (mavlink_commands.cpp:215)
Setting rate failed:Timeout
[01:07:45|Error] Error: set param busy timeout: NAV_FT_RS (mavlink_parameters.cpp:413)
[01:07:45|Error] Failed to set NAV_FT_RS: 0.5 (follow_me_impl.cpp:266)

```

shakthi-prashanth-m, 2018-04-09:
After you switch to `develop`, you've to:
```
git submodule update --init --recursive
make clean
make default
sudo make default install
sudo ldconfig
```
Goto example and re-build.

>[01:07:43|Error] device with UUID: 2 not found (dronecore_impl.cpp:178)

If UUID exists, you've to pass UUID.

julianoes, 2018-04-09:
@varunvp when you have troubles checking out git submodules, also do:
```
git submodule sync --recursive
```

julianoes, 2018-04-09:
The fact that different vehicles in SITL have the same UUID is for sure a problem but I'd say that it should be fixed on the firmware side.

varunvp, 2018-04-09:
> The fact that different vehicles in SITL have the same UUID is for sure a problem but I'd say that it should be fixed on the firmware side.

This is the problem. All the vehicles start with same UUID. Any work around for this, for now? We do have the seperate system id's...

@shakthi-prashanth-m I did all that. I'm modifying my script for the new syntax.

varunvp, 2018-04-09:
OK. I've been able to control all of the drones individually, by changing the UDP ports. Now, I have to see how I should connect to multiple drones and control them.

varunvp, 2018-04-09:
I could control all 4 drones simultaneously! I had to create 4 seperate DroneCore objects, and assign each object one UDP port. Then it worked. So now I have 4 times of some variables required for each drone. Have to put them in an array,I guess. Is this an elegant solution? My worry is tthat 4 DroneCore objects could take a lot of memory, especially if we're running this code on an OffBoard computer. 

On a side note, it would be nice if we had a 'Swarm' type, to which we could add drones, and iterate through them to control them individually, or as a swarm.

julianoes, 2018-04-09:
@varunvp if you want to get this to work with one DroneCore instance, I think you need dive into the Firmware code first and fix it.

> My worry is tthat 4 DroneCore objects could take a lot of memory, especially if we're running this code on an OffBoard computer.

It always makes sense to profile something like that before optimizing. It could well be that it is absolutely not a problem and any kind of optimization would be a waste.

This said, I initially planned DroneCore to be able to handle this case, so multiple vehicles with one DroneCore instance.

varunvp, 2018-04-09:
> if you want to get this to work with one DroneCore instance, I think you need dive into the Firmware code first and fix it.
Sure! Tell me what to change and where ;-) Jokes aside, fixing a professionally maintained repo is out of my skillset for now. I'm a C++ newbie, have a lot to learn. I'll try raising a feature request in PX4 for the time being.

Since this problem has been "solved", I'm going to close it.

closed 2018-04-09 by varunvp

julianoes, 2018-04-10:
Sure, thanks @varunvp!

hamishwillee, 2018-04-10:
@julianoes 

> This said, I initially planned DroneCore to be able to handle this case, so multiple vehicles with one DroneCore instance.

Have we/can we have a test case for this? It should work, but if it doesn't we need to know for sure.

TheNishantK, 2018-04-14:
@varunvp could I have your mail-ID, need to contact you. Or can you ping me on nishantnitk14@gmail.com.
Thanks in advance.

julianoes, 2018-04-17:
> > This said, I initially planned DroneCore to be able to handle this case, so multiple vehicles with one DroneCore instance.
> 
> Have we/can we have a test case for this? It should work, but if it doesn't we need to know for sure.

Yes, at some point we should tackle that.

julianoes, 2018-04-18:
Thanks @hamishwillee for the issue. I agree it is a good selling point but not something immediately required for the projects that I'm working at.