mavlink/MAVSDK #358: Its time to switch Coding style to Clang format

shakthi-prashanth-m, 2018-04-10:
Clang format is most popular C++ coding style: http://clang.llvm.org/docs/ClangFormat.html
Benefits:
- We can also integrate this in git using [git-clang-format](https://github.com/llvm-mirror/clang/blob/master/tools/clang-format/git-clang-format) or add git pre commit hook.
- Clang has a [static analyzer](https://clang-analyzer.llvm.org/).
- It will rearrange header file inclusion order alphabetically (@JonasVautherin was this earlier).
- Supports various styles.
  - LLVM A style complying with the LLVM coding standards
  - Google A style complying with Google’s C++ style guide
  - Chromium A style complying with Chromium’s style guide
  - Mozilla A style complying with Mozilla’s style guide
  - WebKit A style complying with WebKit’s style guide

Reference:
http://clang.llvm.org/docs/ClangFormat.html
http://clang.llvm.org/docs/ClangFormatStyleOptions.html
https://clang-analyzer.llvm.org/
https://clangformat.com/ - Interactive guide and builder.

@julianoes @JonasVautherin @hamishwillee .

JonasVautherin, 2018-04-10:
I added a script to run clang-format to the [avoidance](https://github.com/PX4/avoidance/blob/master/tools/fix_style.sh) module last year. Maybe the script can help:

```
#!/bin/bash

STYLE="google"

if [ "$#" -eq 0 ]; then
    echo "Usage: $0 <src_file | dir>"
    echo ""
    echo "ERROR: At least one source file or one directory must be provided!"

    exit 1
fi

for arg in "$@"
do
    if [ -f $arg ]; then
        clang-format -i -style=${STYLE} $arg
    elif [ -d $arg ]; then
        find $arg -iname *.h -o -iname *.cpp | xargs clang-format -i -style=${STYLE}
    fi
done
```

One thing, though, is that we probably don't want to refactor the whole project, so we would need to keep our current style unchanged, if possible. What do you think?

shakthi-prashanth-m, 2018-04-10:
Yeah, I am also thinking of the same.
I think, if we've any plan to change at all, sooner the better. We can never do anytime in future as the source code is going to be bigger. As it would just change source code appearance for the better. Right ?

JonasVautherin, 2018-04-10:
What I mean is that instead of going for, say, the "google" style, maybe we want to make a custom style that matches (most of) our current style. Does that make sense?

julianoes, 2018-04-12:
> What I mean is that instead of going for, say, the "google" style, maybe we want to make a custom style that matches (most of) our current style. Does that make sense?

Agreed.

PR #410 (2018-05-30): Switch to clang-format
While the clang-format style tries to reflect the previous astyle style
as close as possible, there is, unfortuntately, still a considerable diff.

Fixes #358.
closed 2018-05-30 by julianoes