mavlink/MAVSDK #364: using dronecore with multiple vehicle simlation in ros gazebo

TheNishantK, 2018-04-13:
I am trying to connect to multiple simulated vehicles in ros gaebo simulation using this guide [https://dev.px4.io/en/simulation/multi-vehicle-simulation.html] . But everytime 

> [01:05:53|Error] bind error: Address already in use (udp_connection.cpp:89)
Connection failed: Bind error
Error pops out.

julianoes, 2018-04-13:
@TheNishantK the different vehicles need to use different UDP ports otherwise we get these bind errors.
Have you looked at seen #353?

TheNishantK, 2018-04-14:
I have created following launch file 
`<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav1">
                
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14557"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="rcS" value="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)_$(arg ID)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(arg ID)"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>

    
    <group ns="uav2">
        
        <arg name="ID" value="2"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14559"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="-1"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="rcS" value="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)_$(arg ID)"/>
            <arg name="mavlink_udp_port" value="14562"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
[... 75 lines cut ...]
            <arg name="mavlink_udp_port" value="14568"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(arg ID)"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
</launch>`

and created respective iris files in /posix-configs/SITL/init/ekf2 folder.

and to run the simulation I run following commands.

`git submodule update --init --recursive
make posix_sitl_default
make posix_sitl_default sitl_gazebo
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/sitl_gazebo
roslaunch px4 4uav.launch`

and then i run the dronecore example for takeoff and land for testing, but the bind error pops up.

julianoes, 2018-04-17:
> and then i run the dronecore example for takeoff and land for testing, but the bind error pops up.

What's the exact output?

TheNishantK, 2018-04-18:
nishant@Nishant:~/DroneCore/example/takeoff_land/build$ ./takeoff_and_land 
[04:56:10|Error] bind error: Address already in use (udp_connection.cpp:89)
Connection failed: Bind error

julianoes, 2018-04-18:
@TheNishantK oh I have an idea. You're already using MAVROS in this case right? The problem is that DroneCore uses the same UDP port that MAVROS connects to. You could switch the DroneCore port to that one of QGroundControl if you don't need that one (14550). Otherwise, I don't recommend using DroneCore and MAVROS at the same time.

TheNishantK, 2018-04-20:
The trouble is all of those ports like 14540,14550,14556 all are in use. I have tried all ports, but same error comes out in all cases.

shakthi-prashanth-m, 2018-04-20:
@TheNishantK I don't think you understood https://github.com/dronecore/DroneCore/issues/364#issuecomment-382369537. 
QGroundControl binds to port 14550. So you can't bind to the same port in DroneCore while QGroundControl is running.
MAVROS binds to port 14540. So you can't bind to the same port in DroneCore while MAVROS is running.
What you can do:

- Close QGroundControl and launch DroneCore app using 14550.
- And then, launch MAVROS (which binds to 14540).

closed 2018-07-14 by TheNishantK

Ahmad-Drak, 2019-07-10:
@shakthi-prashanth-m what if i want to run all three together? qgroundcontrol, mavros and mavsdk (dronecore) app. Is there another port which can be connected to? Thanks.

julianoes, 2019-07-10:
@Ahmad-Drak for SITL you could add another `mavlink start` instance here with yet another port:
https://github.com/PX4/Firmware/blob/7ace66a2b9b7ddc91f1f8460a168ebf5a397b050/ROMFS/px4fmu_common/init.d-posix/rcS#L248-L249

I have never tried it but in theory it should work. The only limitation is that we only allow up to 3 mavlink instances but that's for no real reason, it was just arbitrarily chosen at some point.

An alternative is to use mavlink-router to split mavlink traffic to multiple ports.

JonasVautherin, 2019-07-10:
Note that running mavlink-router is super easy ([see here](https://github.com/intel/mavlink-router/#running)).

Ahmad-Drak, 2019-07-18:
@julianoes this sounds more complicated than the mavlink-router, wouldn't want to mess with rcs
@JonasVautherin I tried using the mavlink-router, ran the command 
`mavlink-routerd -e 127.0.0.1:14552 0.0.0.0:24550`

Mavros is connected on 14540
qgroundcontrol is connected on 14550

Then tried to run a MAVSDK example as
`./follow_me udp://:14552`

buit all i got was 'wait for system to connect via heartbeat'..Was the mavlink-router command not correct? shouldn't  it bind to 14552?

JonasVautherin, 2019-07-18:
Are you sure about `0.0.0.0:24550`? I will listen on that, and send to the endpoint (`-e 127.0.0.1:14552`).

Ahmad-Drak, 2019-07-18:
No not really. It was suggested [here](https://github.com/intel/mavlink-router)

> It's also possible to route mavlinks packets from any interface using:

> $ mavlink-routerd -e 192.168.7.1:14550 -e 127.0.0.1:14550  0.0.0.0:24550

I tired different combinations for the udp_address according to the below

mavlink-routerd [OPTIONS...] [<uart>|<udp_address>]

JonasVautherin, 2019-07-18:
Right. What are you trying to do exactly? Multiple vehicles in simulation, or just one, but that you want to connect to from QGC, MAVSDK and MAVROS at the same time?

JonasVautherin, 2019-07-18:
You could for instance keep 14550 untouched for QGC, and forward 14540 to, say, 14541 and 14542 for MAVROS and MAVSDK:

```sh
mavlink-routerd -e 127.0.0.1:14541 -e 127.0.0.1:14542 0.0.0.0:14540
```

To summarize:

* QGC takes 14550
* mavlink-router takes 14540
* MAVROS takes 14541 (created by mavlink-router)
* MAVSDK takes 14542 (created by mavlink-router)

JonasVautherin, 2019-07-20:
Does that work then?

Ahmad-Drak, 2019-07-20:
Yes works perfectly. Thank you.