mavlink/MAVSDK #367: mavlink 1.0 is not supported?

Seunghwan17, 2018-04-20:
mavlink v1.0 is not supported?
I use a esp8266 which has firmware mavesp8266.
But, DroneCore has mavlink v2.0.
Anyone have similar problem?

Seunghwan17, 2018-04-20:
@julianoes 

um... I use a firmware mavesp8266.
It just use mavlink v1.0...
Do you know how to upgarde mavlink v1.0 -> mavlink v2.0?
this firmware inclue mavlink v1.0 (c_library_v1)

https://github.com/dogmaphobic/mavesp8266

julianoes, 2018-04-20:
@Seunghwan17 it looks like the mavesp8266 firmware doesn't support mavlink v2.0 yet:
https://github.com/dogmaphobic/mavesp8266/issues/45

julianoes, 2018-04-23:
@Seunghwan17 Similar to the mavesp8266 would be solutions on 433 MHz or 900 MHz: https://www.amazon.com/433-European-Telemetry-Radio-Vehicle/dp/B06ZZN4QKW. These just transmit serial data and therefore should not care about MAVLink 1 vs 2.

However, the right thing would be to add MAVLink 2 support to the mavesp8266 firmware as noted in https://github.com/dogmaphobic/mavesp8266/issues/45.

hamishwillee, 2018-05-15:
Suggest we close this. I have added FAQ item that MAVLink 1 is not supported: https://github.com/Dronecode/sdk_docs/pull/120

closed 2018-05-15 by julianoes