mavlink/MAVSDK #381: fly_qgc_mission error loading shared libraries: libjson11.so

hamishwillee, 2018-05-03:
```
/DroneCore/example/fly_qgc_mission/build$ ./fly_qgc_mission 
./fly_qgc_mission: error while loading shared libraries: libjson11.so: cannot open shared object file: No such file or directory
``

julianoes, 2018-05-03:
Thx @hamishwillee

julianoes, 2018-05-16:
@hamishwillee I just tried this and it works for me on Fedora 28 after I do this:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
sudo ldconfig
```

julianoes, 2018-05-16:
But building this example on macOS fails:
```
make
Scanning dependencies of target fly_qgc_mission
[ 50%] Building CXX object CMakeFiles/fly_qgc_mission.dir/fly_qgc_mission.cpp.o
[100%] Linking CXX executable fly_qgc_mission
Undefined symbols for architecture x86_64:
  "json11::Json::parse(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&, json11::JsonParse)", referenced from:
      dronecore::MissionImpl::import_qgroundcontrol_mission(std::__1::vector<std::__1::shared_ptr<dronecore::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecore::MissionItem> > >&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in libdronecore_mission.a(mission_impl.cpp.o)
  "json11::Json::array_items() const", referenced from:
      dronecore::MissionImpl::import_mission_items(std::__1::vector<std::__1::shared_ptr<dronecore::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecore::MissionItem> > >&, json11::Json const&) in libdronecore_mission.a(mission_impl.cpp.o)
  "json11::Json::number_value() const", referenced from:
      dronecore::MissionImpl::import_mission_items(std::__1::vector<std::__1::shared_ptr<dronecore::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecore::MissionItem> > >&, json11::Json const&) in libdronecore_mission.a(mission_impl.cpp.o)
  "json11::Json::int_value() const", referenced from:
      dronecore::MissionImpl::import_mission_items(std::__1::vector<std::__1::shared_ptr<dronecore::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecore::MissionItem> > >&, json11::Json const&) in libdronecore_mission.a(mission_impl.cpp.o)
  "json11::Json::operator[](std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      dronecore::MissionImpl::import_mission_items(std::__1::vector<std::__1::shared_ptr<dronecore::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecore::MissionItem> > >&, json11::Json const&) in libdronecore_mission.a(mission_impl.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [fly_qgc_mission] Error 1
make[1]: *** [CMakeFiles/fly_qgc_mission.dir/all] Error 2
make: *** [all] Error 2
```

PR #395 (2018-05-17): fly_qgc_mission: link to json11
We need to link to libjson11 as well because libdronecore_mission.so doesn't include it.

@JonasVautherin any better idea? Or should we include json11 inside `libdronecore.a` like tinyxml2 to prevent this?

Fixes #381.
closed 2018-05-17 by julianoes