mavlink/MAVSDK #399: fly_mission freezes on mission accepted

hamishwillee, 2018-05-18:
The future does not appear to be returning with mission uploaded.

```
hamish@hamish-ThinkPad-T470:~/github/dronecore/2test/DroneCore/example/fly_mission/build$ ./fly_mission udp://:14540
Waiting to discover system...
[04:48:15|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:208)
[04:48:15|Debug] New: System ID: 1 Comp ID: 1 (dronecore_impl.cpp:286)
[04:48:15|Debug] Component Autopilot added. (mavlink_system.cpp:330)
[04:48:15|Debug] Found 1 component(s). (mavlink_system.cpp:462)
[04:48:15|Debug] Discovered 4294967298 (mavlink_system.cpp:464)
Discovered system with UUID: 4294967298
Waiting for system to be ready
[04:48:15|Debug] We got an ack: 520 (mavlink_commands.cpp:140)
[04:48:16|Debug] We got an ack: 511 (mavlink_commands.cpp:140)
System ready
Creating and uploading mission
Uploading mission...
[04:48:16|Debug] Send mission item 0 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 1 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 2 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 3 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 4 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 5 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 6 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 7 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 8 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 9 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 10 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 11 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 12 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 13 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 14 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 15 (mission_impl.cpp:904)
[04:48:16|Debug] Send mission item 16 (mission_impl.cpp:904)
[04:48:17|Debug] Send mission item 17 (mission_impl.cpp:904)
[04:48:17|Debug] Send mission item 18 (mission_impl.cpp:904)
[04:48:17|Debug] Send mission item 19 (mission_impl.cpp:904)
[04:48:17|Debug] Send mission item 20 (mission_impl.cpp:904)
[04:48:17|Debug] Send mission item 21 (mission_impl.cpp:904)
[04:48:17|Debug] Send mission item 22 (mission_impl.cpp:904)
[04:48:17|Info ] Mission accepted (mission_impl.cpp:162)
```

julianoes, 2018-05-18:
@hamishwillee is this the latest latest? I just tested this yesterday and it worked multiple times.

hamishwillee, 2018-05-19:
Yes, it was the latest latest, against the latest latest PX4 firmware. The other examples worked fine.

julianoes, 2018-05-19:
Oh, that's the example and not the integration test. Sorry, I'll test that one then next.

julianoes, 2018-05-25:
I'm running into another problem:

```
Waiting for system to be ready
System ready
Creating and uploading mission
Uploading mission...
[11:03:21|Warn ] Mission handling timed out while uploading mission. (mission_impl.cpp:1058)
```

Maybe something with the mission handling changed (broke?) in the firmware.

julianoes, 2018-05-25:
Nevermind, the integration test works, so this is something about the example.

julianoes, 2018-05-25:
So, the example works just fine on my Fedora Linux but stalls on macOS.

@hamishwillee when I ask latest latest, I mean: have you made sure that you have built and installed the latest?

julianoes, 2018-05-25:
Ok, after debugging this for several hours, I found that I had leftover library files in `/usr/local/lib/`.

Once I cleaned it up, did make clean, it all worked fine.

```
sudo rm -rf "/usr/local/lib/libdrone*"
make clean
make default
sudo make default install
(cd example/fly_mission && rm -rf build && mkdir build && (cd build && cmake .. && make && ./fly_mission udp://))
```

hamishwillee, 2018-05-28:
>  latest latest, I mean: have you made sure that you have built and installed the latest?

I would have said yes, but when I tried this again with a clean build, it is indeed working. I didn't have to delete "/usr/local/lib/libdrone*". 

I almost wish it hadn't worked so I would feel like I hadn't wasted your time. Sorry!

closed 2018-05-28 by hamishwillee

julianoes, 2018-05-29:
No worries, thanks @hamishwillee!