mavlink/MAVSDK #411: what is DroneCores's uuid with multiple vehicles?

Seunghwan17, 2018-05-25:
I don't understand UUID is  became AUTOPILOT_VERSION's uid.

To explain my situation I have five drones.
According to the example, when dronecore make a system that need UUID. 

like below
DroneCore dc;
//... 
//Use UUID named uuid to get the associated System
System &system = dc.system(uuid);

But in dronecore's UUID is declared AUTOPILOT_VERSION's uid. My five drones have same AUTOPILOT_VERSION's uid and have different system ID. So, my five drones collision each others. 
I have question, my AUTOPILOT_VERSION's uid are same is strange or dronecore's uuid is strange?

I would be grateful if you could help with this phenomenon.

julianoes, 2018-05-25:
Thanks for the question. In my opinion, the issue here is that each drone simulated by SITL has the same UUID. UUID stands "Universally unique identifier", which means they should be unique 😄 .

One workaround for you would be to add something to the UUID in each of your instances, so that they differ from each other.

And note that you also need to set different system IDs for each of the drones.

Seunghwan17, 2018-05-25:
Um.. in my case I don't use SITL.
My five drones in real case.  My five different drones have same UUID.....
Maybe it is idiot question do you have many drones?

julianoes, 2018-05-25:
> Um.. in my case I don't use SITL.

Oh, sorry. I was just assuming that. Ok, so what PX4 version are you using on them?
And could the issue be that DroneCore is still using the field `uid` instead of `uid2` in http://mavlink.org/messages/common#AUTOPILOT_VERSION? You can check if the `uid2` field is actually different.
We are planning to move to `uid2` in the future.

>  Maybe it is idiot question do you have many drones?

Good question 😄 . So I have a couple of drones but I have not used DroneCore with multiple drones at the same time. We've implemented it in a way to potentially support more than one but there might be some work left to get it to work properly.

Seunghwan17, 2018-05-25:
[image]
[image]

this is my two drones.
these two drones have same uid. It is my  wondering. The situation that two drones have same value is  normal?

below two pictures are my drones version

[image]
[image]

Seunghwan17, 2018-05-25:
Any my drones have PixRacer. AUTOPILOT_VERSION's uid and uid2 are same. But, ver'UIDs are different.

julianoes, 2018-05-25:
That's odd, the uid and uid2 should be taken from the STM32 UUID of the chip, so that shouldn't happen.

I suggest you create an issue in PX4/Firmware and add all the detail for the two Pixracers.

Seunghwan17, 2018-05-25:
Thanks. I create an issue in PX4/Firmware.
What isUUID's role in dronecore? It is important?
If is not, I try to replace UUID to system ID.

julianoes, 2018-05-25:
yer, system ID should work too. Thea idea was to use UUID to be able to refer to individual vehicles over time and not just per session but that's currently not used yet anyway.

mzahana, 2018-08-11:
@Seunghwan17 Did you manage to use Dronecode SDK with multiple drones?

I am trying to do the same, and I actually would like to use system ID (MAVLink ID which can be set by QGC), but I don't even see it available in the [System class](https://sdk.dronecode.org/en/api_reference/classdronecode__sdk_1_1_system.html)

@julianoes Is the MAVLink ID provided some how by the SDK?

Thanks.

julianoes, 2018-08-13:
> @julianoes Is the MAVLink ID provided some how by the SDK?

No, only the UUID for now. Let's close this because the PX4 Firmware issue has been fixed and follow up in #486.

closed 2018-08-13 by julianoes

Seunghwan17, 2018-08-13:
@mzahana oops.. I missed your message. In my case, I edit dronecode SDK to use sys ID not a UUID. As you see this post, UUID has some problem to me. If you want, I will show this code.

julianoes, 2018-08-13:
Could you describe the problem that you are facing?

mzahana, 2018-08-13:
I am working with a set of ordered drones. They have unique NAVLink IDs. So, it is easier for me to deal with system IDs than UUIDs.

mzahana, 2018-08-13:
@Seunghwan17 I saw that system_id is available [here](https://github.com/Dronecode/DronecodeSDK/blob/develop/core/system_impl.cpp#L518)

The discussion is moved to https://github.com/Dronecode/DronecodeSDK/issues/486