mavlink/MAVSDK #417: v1.0.0 Changelog scratchpad

mrpollo, 2018-06-05:
Hey everyone I generated the following changelog, can we please review?

---

# Change Log

## [v1.0.0](https://github.com/dronecore/DroneCore/tree/v1.0.0) (2018-06-04)

[Full Changelog](https://github.com/dronecore/DroneCore/compare/v0.5.0...v1.0.0)

**Closed issues:**

- Add documentation for methods in device.h [\#296](https://github.com/dronecore/DroneCore/issues/296)

**Merged pull requests:**

- Public comments improvements [\#415](https://github.com/dronecore/DroneCore/pull/415) ([julianoes](https://github.com/julianoes))

## [v0.5.0](https://github.com/dronecore/DroneCore/tree/v0.5.0) (2018-06-04)

[Full Changelog](https://github.com/dronecore/DroneCore/compare/v0.4.0...v0.5.0)

**Implemented enhancements:**

- Python bindings for the core library [\#388](https://github.com/dronecore/DroneCore/issues/388)
- Examples should use consistent connection startup semantics [\#380](https://github.com/dronecore/DroneCore/issues/380)
- Implement mission features in backend [\#350](https://github.com/dronecore/DroneCore/issues/350)

**Fixed bugs:**

- transition\_vtol\_fixed\_wing example never ready to arm [\#401](https://github.com/dronecore/DroneCore/issues/401)
- fly\_qgc\_mission error loading shared libraries: libjson11.so [\#381](https://github.com/dronecore/DroneCore/issues/381)
- Mission subscribe blocked by mission pause [\#373](https://github.com/dronecore/DroneCore/issues/373)
- Random crash in Yuneec SDK [\#309](https://github.com/dronecore/DroneCore/issues/309)

**Closed issues:**

- fly\_mission and fly\_qgc\_mission examples freezes on mission accepted [\#399](https://github.com/dronecore/DroneCore/issues/399)
- \[Help\]  How to use DroneCore in HITL Gazebo [\#355](https://github.com/dronecore/DroneCore/issues/355)
- Implement remaining actions in backend [\#315](https://github.com/dronecore/DroneCore/issues/315)

**Merged pull requests:**

- Implement mission backend [\#416](https://github.com/dronecore/DroneCore/pull/416) ([JonasVautherin](https://github.com/JonasVautherin))
- camera: add set\_mode \(synchronous\) [\#414](https://github.com/dronecore/DroneCore/pull/414) ([JonasVautherin](https://github.com/JonasVautherin))
- core: fix macOS warning [\#413](https://github.com/dronecore/DroneCore/pull/413) ([julianoes](https://github.com/julianoes))
- Fix clang format [\#412](https://github.com/dronecore/DroneCore/pull/412) ([JonasVautherin](https://github.com/JonasVautherin))
- Switch to clang-format [\#410](https://github.com/dronecore/DroneCore/pull/410) ([julianoes](https://github.com/julianoes))
- Implement camera backend [\#409](https://github.com/dronecore/DroneCore/pull/409) ([JonasVautherin](https://github.com/JonasVautherin))
- backend: update proto and add camera [\#407](https://github.com/dronecore/DroneCore/pull/407) ([JonasVautherin](https://github.com/JonasVautherin))
- System refactor [\#406](https://github.com/dronecore/DroneCore/pull/406) ([julianoes](https://github.com/julianoes))
- Mission pause [\#405](https://github.com/dronecore/DroneCore/pull/405) ([Sushma7785](https://github.com/Sushma7785))
- Fix VTOL example [\#403](https://github.com/dronecore/DroneCore/pull/403) ([julianoes](https://github.com/julianoes))
- mission: fix warning about missing return [\#402](https://github.com/dronecore/DroneCore/pull/402) ([julianoes](https://github.com/julianoes))
- Make all examples use add\_any\_connection\(\) [\#400](https://github.com/dronecore/DroneCore/pull/400) ([hamishwillee](https://github.com/hamishwillee))
- Various fixes found doing camera tests [\#398](https://github.com/dronecore/DroneCore/pull/398) ([julianoes](https://github.com/julianoes))
- backend: add mission finished [\#397](https://github.com/dronecore/DroneCore/pull/397) ([JonasVautherin](https://github.com/JonasVautherin))
- Add thread pool [\#396](https://github.com/dronecore/DroneCore/pull/396) ([julianoes](https://github.com/julianoes))
- fly\_qgc\_mission: link to json11 [\#395](https://github.com/dronecore/DroneCore/pull/395) ([julianoes](https://github.com/julianoes))
- Fix connection comments [\#394](https://github.com/dronecore/DroneCore/pull/394) ([julianoes](https://github.com/julianoes))
- Fix timeout handler segfault [\#393](https://github.com/dronecore/DroneCore/pull/393) ([julianoes](https://github.com/julianoes))
- build\_system: remove build for armeabi [\#392](https://github.com/dronecore/DroneCore/pull/392) ([julianoes](https://github.com/julianoes))
- mavlink command fixes [\#391](https://github.com/dronecore/DroneCore/pull/391) ([julianoes](https://github.com/julianoes))
- core: use 251 as compid [\#390](https://github.com/dronecore/DroneCore/pull/390) ([julianoes](https://github.com/julianoes))
- Fix Mission lockup [\#387](https://github.com/dronecore/DroneCore/pull/387) ([julianoes](https://github.com/julianoes))
- build\_system: build tinyxml2 statically [\#386](https://github.com/dronecore/DroneCore/pull/386) ([JonasVautherin](https://github.com/JonasVautherin))
- mission: use latest commit of json11's fork [\#385](https://github.com/dronecore/DroneCore/pull/385) ([JonasVautherin](https://github.com/JonasVautherin))
- core: fix bad-cast on param\_get fail [\#384](https://github.com/dronecore/DroneCore/pull/384) ([julianoes](https://github.com/julianoes))
- core: fix conversion that now generates a warning [\#383](https://github.com/dronecore/DroneCore/pull/383) ([JonasVautherin](https://github.com/JonasVautherin))
- backend: fix wrong unit test in mission [\#382](https://github.com/dronecore/DroneCore/pull/382) ([JonasVautherin](https://github.com/JonasVautherin))
[... 131 lines cut ...]
- Fix docker builds [\#255](https://github.com/dronecore/DroneCore/pull/255) ([julianoes](https://github.com/julianoes))
- cmake: add tinyxml2 install hack for Mac [\#253](https://github.com/dronecore/DroneCore/pull/253) ([julianoes](https://github.com/julianoes))
- cmake: install tinyxml2 library for Android [\#252](https://github.com/dronecore/DroneCore/pull/252) ([julianoes](https://github.com/julianoes))
- Makefile: fix make clean after make android [\#251](https://github.com/dronecore/DroneCore/pull/251) ([julianoes](https://github.com/julianoes))
- Fix offboard plugin creation in example [\#246](https://github.com/dronecore/DroneCore/pull/246) ([hamishwillee](https://github.com/hamishwillee))
- Fix up plugin constructor docs [\#242](https://github.com/dronecore/DroneCore/pull/242) ([hamishwillee](https://github.com/hamishwillee))
- build\_system: fix android build and make curl include more consistent [\#241](https://github.com/dronecore/DroneCore/pull/241) ([JonasVautherin](https://github.com/JonasVautherin))
- Enable external plugins again [\#239](https://github.com/dronecore/DroneCore/pull/239) ([julianoes](https://github.com/julianoes))
- Fix `generate\_doc.sh` issue \(\#236\) [\#237](https://github.com/dronecore/DroneCore/pull/237) ([shakthi-prashanth-m](https://github.com/shakthi-prashanth-m))
- Add QGC Mission import capability [\#235](https://github.com/dronecore/DroneCore/pull/235) ([shakthi-prashanth-m](https://github.com/shakthi-prashanth-m))
- Robustify mission upload/download [\#230](https://github.com/dronecore/DroneCore/pull/230) ([julianoes](https://github.com/julianoes))
- Fix issue: https://github.com/dronecore/DroneCore/issues/227 [\#228](https://github.com/dronecore/DroneCore/pull/228) ([shakthi-prashanth-m](https://github.com/shakthi-prashanth-m))
- mavlink: update submodule [\#225](https://github.com/dronecore/DroneCore/pull/225) ([julianoes](https://github.com/julianoes))
- Correct API description of Telemetry and Offboard plugins. [\#223](https://github.com/dronecore/DroneCore/pull/223) ([shakthi-prashanth-m](https://github.com/shakthi-prashanth-m))
- curl-android-ios: update for openssl headers [\#220](https://github.com/dronecore/DroneCore/pull/220) ([julianoes](https://github.com/julianoes))
- curl-android-ios: swap to dronecore fork [\#218](https://github.com/dronecore/DroneCore/pull/218) ([julianoes](https://github.com/julianoes))
- Keep previous uuid when Device disconnects [\#215](https://github.com/dronecore/DroneCore/pull/215) ([fklabach](https://github.com/fklabach))
- Set Flight Mode using new interface set\_flight\_mode in DeviceImpl [\#206](https://github.com/dronecore/DroneCore/pull/206) ([Rjasuja](https://github.com/Rjasuja))
- core: try to reconnect TCP connection [\#203](https://github.com/dronecore/DroneCore/pull/203) ([julianoes](https://github.com/julianoes))
[... cut ...]
- Package integration test, generate .deb and .rpm [\#198](https://github.com/dronecore/DroneCore/pull/198) ([julianoes](https://github.com/julianoes))
- grpc: reorganize grpc [\#197](https://github.com/dronecore/DroneCore/pull/197) ([JonasVautherin](https://github.com/JonasVautherin))
- More flexible plugin architecture [\#188](https://github.com/dronecore/DroneCore/pull/188) ([julianoes](https://github.com/julianoes))

\* *This Change Log was automatically generated by [github_changelog_generator](https://github.com/skywinder/Github-Changelog-Generator)*

julianoes, 2018-06-05:
@mrpollo that's nice to auto-generate it but I actually wrote the changelogs manually in the release notes, check out https://github.com/dronecore/DroneCore/releases. This way it's easier to read and less verbose.

hamishwillee, 2018-06-05:
@julianoes The docs release happen after the fact, but should match your releases: https://github.com/dronecore/sdk_docs/releases
At the top is a link to associated docs. It might be worth creating just that link to the online version of the docs in the software release notes (?)

anitha-suresh, 2018-06-06:
@mrpollo looks good! As it is auto generated, it would be effortless. I agree with Julian, that it looks too verbose. Can we please not categorize 'Fixed', 'Closed', 'Merged'. 

In release notes, it is easier and makes sense to mention at a higher level, about what feature or functionality is enabled (as there might be some 10 items/subtasks going as part of enabling one feature/function) AND also on the major bug fixes.

These are my thoughts :) Will wait for others to comment

JonasVautherin, 2018-06-06:
> Can we please not categorize 'Fixed', 'Closed', 'Merged'.

There is a difference between those sections, but I believe the keywords are more: 'enhancements', 'bugs', 'issues', 'pull requests'. I guess the first two come from the labels. For the last two, the thing is that some PRs are not related to an issue.

If you want to simplify that, maybe we could just have 'bugs' and 'enhancements' (e.g. everything that is not labeled as a bug is an enhancement). And one question is whether we need to list both issues and PRs, or if we can keep only the PRs (because most of the time, issues are closed by PRs, but some PRs are not related to an issue). But then it means that we should make sure we put the 'bug' label on the PRs themselves.

@mrpollo I like the generated report! That is neat for the "full changes" report. And at the top we should probably highlight a few features/write a high-level description of the changes.

Gradle does it this way: they have some kind of blog post that says "now with gradle X.Y, you can do this, this, this and that!", then they have detailed explanations for those "key" features (sometimes linking to actual examples), and I believe they have a link to the full changelog somewhere.

anitha-suresh, 2018-06-11:
Can we please not categorize 'Fixed', 'Closed', 'Merged' - I meant do we need all three as part of release notes.

julianoes, 2019-01-09:
I'll continue doing this manually for now.

closed 2019-01-09 by julianoes