mavlink/MAVSDK #424: Dronecore as subproject

Stifael, 2018-06-11:
If dronecore is used as a subproject, in order to compile correctly with the correct paths, each plugin and core requires `target_include_directories(dronecore PUBLIC ${CMAKE_CURRENT_SOURCE_DIR})`, otherwise the header files will not be found.

julianoes, 2018-06-11:
This has the drawback that also internal header files within a plugin would be public.
E.g. for the action plugin, `action.h` should be public but `action_impl.h` should not.

Stifael, 2018-06-19:
ok.
Have you ever tried to run one of the examples without having it installed on the system?

Stifael, 2018-06-19:
for instance: https://github.com/dronecore/DroneCore/blob/develop/example/offboard_velocity/offboard_velocity.cpp#L15 only works when it is installed on the system.

julianoes, 2018-06-25:
Another way would be to install the plugin header files in a `plugins` subdirectory. That way the installed directory structure would be closer to the non-installed one.

The change would be to install it as:
```
dronecore/plugins/action/action.h
```
instead of:
```
dronecore/action.h
```

julianoes, 2018-08-14:
@Stifael is this still an issue or can this get closed?

Stifael, 2018-08-15:
no issue. can be closed.

closed 2018-08-15 by Stifael