mavlink/MAVSDK #429: Decide how to pass callbacks to API

julianoes, 2018-06-14:
I see that we have inconsistency in the way callbacks are passed into the C++ API.

Some of the APIs use pass-by-value:
```
void subscribe_plugin_something(const subscribe_plugin_something_callback_t callback);
```

Others use pass-by-reference:
```
void subscribe_plugin_something(const subscribe_plugin_something_callback_t &callback);
```

From my understanding, pass-by-value leads to a copy (shallow or deep?) of the function object and therefore adds overhead, however, it prevents the case where the passed callback goes out of scope which can lead to a segfault.

@shakthi-prashanth-m @JonasVautherin opinions?

shakthi-prashanth-m, 2018-06-14:
>passed callback goes out of scope which can lead to a segfault.

I dont think this happens, because callback is simply pointer to a function which exists thought the execution of the application (unless it is being loaded dynamically using `dlopen`). IMO, pass by value is ok.

JonasVautherin, 2018-06-14:
I would pass by value as well. I found [this answer](https://stackoverflow.com/a/18366273/1368342) very interesting, including:

> That store makes passing by value more optimal. If there is any possibility you are storing a copy of the std::function, pass by value. Otherwise, either way is roughly equivalent: the only downside to by-value is if you are taking the same bulky std::function and having one sub method after another use it. Barring that, a move will be as efficient as a const&.

julianoes, 2018-06-14:
Do we have to use `std::move` anywhere or does that happen automatically?

JonasVautherin, 2018-06-15:
From what I understand, it happens automatically. Otherwise it would really not be practical...

julianoes, 2024-03-04:
This has changed in the meantime anyway, and it seems to work well enough.

closed 2024-03-04 by julianoes