mavlink/MAVSDK #452: NDK_ROOT is undefined

canopusTest, 2018-07-17:
I am using the DroneCodeSDK . But when I try to build for the android , I hit this command  "sudo make android install" but I get this error" Makefile:152: *** NDK_ROOT is undefined, please point the environment variable to android-ndk root..  Stop". Although I have set the NDK_ROOT in .bashrc. Also i have set NDK_ROOT from command prompt . I am using Ubuntu. I am not able to build the code for android. Please suggest how to get rid of this error.

julianoes, 2018-07-18:
@canopusTest is the env variable `NDK_ROOT` defined correctly?

Before doing make, try: `echo $NDK_ROOT` and check if the path is correct by doing `ls $NDK_ROOT`.
To manually set it to try out, do:
`export $NDK_ROOT=/path/to/ndk/root`

canopusTest, 2018-07-19:
Yes , It is set to correct path and previous error are now fixed. But still I am getting some error while hitting the command "sudo make android install". This is the error " nstall configuration: "Debug"
-- Installing: /usr/local/lib/android/x86/libdronecode_sdk.so
CMake Error at core/cmake_install.cmake:36 (file):
  file INSTALL cannot copy file
  "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/core/libdronecode_sdk.so"
  to "/usr/local/lib/android/x86/libdronecode_sdk.so".
Call Stack (most recent call first):
  cmake_install.cmake:38 (include)".
Please help me out to get the code compiled.

canopusTest, 2018-07-19:
There are two things happening:- 
Case 1). When I run command as "sudo make android install". Then I get the NDK_ROOT error. 
 I have set the paths as below 
 export NDK_ROOT=/home/akash/Android/Sdk/ndk-bundle
 export ANDROID_TOOLCHAIN_CMAKE=/home/akash/Android/Sdk/ndk-bundle/build/cmake/android.toolchain.cmake
 export ANDROID_CMAKE_BIN=/home/akash/Android/Sdk/cmake/3.6.4111459/bin/cmake

I have ensured that the paths are set correctly by running echo$NDK_ROOT I have echo all three paths and yet they are pointing to correct location. But still I am getting the error "Makefile:152: *** NDK_ROOT is undefined, please point the environment variable to android-ndk root.. Stop."

Case 2). When I run the command without sudo as "make android install" then I get the following error
" file INSTALL cannot copy file
  "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/core/libdronecode_sdk.so"
  to "/usr/local/lib/android/x86/libdronecode_sdk.so".
Call Stack (most recent call first):
  cmake_install.cmake:38 (include)
Makefile:93: recipe for target 'install' failed
make[1]: *** [install] Error 1
Makefile:86: recipe for target 'android_x86' failed
make: *** [android_x86] Error 2"

Please help me out to get this compiled.

julianoes, 2018-07-19:
@canopusTest aha!

So it looks like you need to specify env variables to pass them on to sudo:
https://stackoverflow.com/questions/8633461/how-to-keep-environment-variables-when-using-sudo

However, you don't need to install the Android artifacts into `/usr/local` of your system assuming you are not compiling this on an Android system. Best is to install them somewhere where you don't need sudo.

My scripts look like this but I have not verified if they still work in a while:
```
make INSTALL_PREFIX=$(pwd)/install BUILD_TYPE=$build_type android install
cp -r DroneCore/core/third_party/curl-android-ios/prebuilt-with-ssl/android/* install/lib/android/
cp -r DroneCore/core/third_party/curl-android-ios/curl/COPYING install/CURL_COPYING
mkdir -p install/include/curl
cp install/lib/android/include/curl/* install/include/curl/
rm -rf install/lib/android/include/
```

canopusTest, 2018-07-20:
Thanks @julianoes  for the help so far. I would like to clearify that I am compiling the code on Ubuntu 16.04. I am following the instructions to build from here "https://sdk.dronecode.org/en/contributing/build.html" . I am compiling the code for an android app. When I use the command " make android install". I get the error Case 2). in above comment. I have checked the script and it looks like below:- 

# Install script for directory: /home/akash/Desktop/DroneCodeRetry/DronecodeSDK

# Install shared libraries without execute permission?
if(NOT DEFINED CMAKE_INSTALL_SO_NO_EXE)
  set(CMAKE_INSTALL_SO_NO_EXE "1")
endif()

if("${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/lib/android/x86" TYPE SHARED_LIBRARY FILES "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/core/libdronecode_sdk.so")
  if(EXISTS "$ENV{DESTDIR}${CMAKE_INSTALL_PREFIX}/lib/android/x86/libdronecode_sdk.so" AND
     NOT IS_SYMLINK "$ENV{DESTDIR}${CMAKE_INSTALL_PREFIX}/lib/android/x86/libdronecode_sdk.so")
    if(CMAKE_INSTALL_DO_STRIP)
      execute_process(COMMAND "/home/akash/Android/Sdk/ndk-bundle/toolchains/x86-4.9/prebuilt/linux-x86_64/bin/i686-linux-android-strip" "$ENV{DESTDIR}${CMAKE_INSTALL_PREFIX}/lib/android/x86/libdronecode_sdk.so")
    endif()
  endif()
endif()

if("${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include/dronecode_sdk" TYPE FILE FILES
    "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/core/connection_result.h"
    "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/core/system.h"
    "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/core/dronecode_sdk.h"
    "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/core/plugin_base.h"
    )
endif()

--------
This is where the code fails the compilation. This script is under the class "cmake_install.cmake" . I am not sure what is wrong in the script. Please let me know the compiling steps and building this code for android.
Thanks

julianoes, 2018-07-20:
Hm, two things:
1. Please put code and console output inside ``` here on github, that makes it easier to read.
2. You need to do `make clean` before you change the install directory, so:
    ```
    make clean # or maybe sudo make clean 
    make INSTALL_PREFIX=$(pwd)/install android install
    ```

canopusTest, 2018-07-20:
I run the commands you mentioned, Below is the console output:- 

```CMake Error at plugins/mission/cmake_install.cmake:47 (file):
  file INSTALL cannot find
  "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/plugins/mission/third_party/json11/libjson11.so".```.

The cmake code from line 47 is as below:- 

```if("${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
        file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/lib/android/x86" TYPE FILE FILES "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/plugins/mission/third_party/json11/libjson11.so")
        endif()

        if(NOT CMAKE_INSTALL_LOCAL_ONLY)
        # Include the install script for each subdirectory.
        include("/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/plugins/mission/third_party/json11/cmake_install.cmake")

        endif()```

Thanks.

julianoes, 2018-07-20:
Nice, I just ran into this as well. Pushing a fix in a second.

julianoes, 2018-07-20:
Please try #456.

canopusTest, 2018-07-20:
Yes , I tried after a pull and now this is the newly introduced error from console output:- 

```Installing: /home/akash/Desktop/DroneCodeRetry/DronecodeSDK/install/lib/android/x86/libdronecode_sdk_mission.a
CMake Error at plugins/mission/cmake_install.cmake:47 (file):
  file INSTALL cannot find
  "/home/akash/Desktop/DroneCodeRetry/DronecodeSDK/build/android_x86/plugins/mission/third_party/json11/libjson11.so".
Call Stack (most recent call first):
  plugins/cmake_install.cmake:39 (include)
  cmake_install.cmake:39 (include)

julianoes, 2018-07-20:
The fix is not merged yet. Please try the branch.

closed 2018-07-23 by julianoes

canopusTest, 2018-07-25:
I do make the changes in the  ``` plugins/mission/CMakeLists.txt ``` file and then I run the below commands:- 

 ```make clean # or maybe sudo make clean ```
 ```make INSTALL_PREFIX=$(pwd)/install android install```

 and it run all fine without any error on the terminal.

 Please let me know to run the android sample app within the DroneCodeSDK , how can I import the code into Android Studio?

 ---
Thanks.

canopusTest, 2018-07-25:
@julianoes Can you please check in, I will get back to you.

canopusTest, 2018-07-27:
@julianoes Do you have any update for the same.

julianoes, 2018-07-30:
Not yet, sorry. I have a bunch of other things on my list.

canopusTest, 2018-07-31:
Okay , Can you please let me know if this issue is only for Android build or this issue exists fir iOS build too. ?

julianoes, 2018-07-31:
We are currently working on iOS, so that's closer to be usable.

canopusTest, 2018-07-31:
Okay , How long will it take to get the iOS build ready ?

julianoes, 2018-07-31:
I'm hoping only about 2 weeks.

julianoes, 2018-08-02:
I this case I suggest you use the C++ API and wrap it yourself manually for now.

canopusTest, 2018-08-04:
Is that the only left option ? This will take even more time.

julianoes, 2018-08-06:
You can try the Dronecode SDK in Swift using Cocoapods, see https://github.com/Dronecode/DronecodeSDK-Swift

PR #456 (2018-07-23): Fix Android build issues
A couple of build issues for Android came up.

Presumably we should add the Android build to CI if we don't already do it.

Fixes #452.