mavlink/MAVSDK #454: Dertermine exact time fly

dangviethieu-hntv, 2018-07-18:
Hi all, I have mission with take photo action each waypoints. How I can determine exact time when drone fly from waypoint to waypoint and time drone do action at waypoint?

julianoes, 2018-07-18:
@dangviethieu-hntv this would depend on the mission speed that you set. Given the distance you could then calculate the approximate time required to fly it.

dangviethieu-hntv, 2018-07-19:
@julianoes How about determine time drone do action at waypoint, sir? (take photo, control gimbal)

hamishwillee, 2018-07-19:
> How about determine time drone do action at waypoint, sir? (take photo, control gimbal)

That would depend on the gimbal and camera you are using. 

Does it matter? You're never going to be able to get a perfect estimate - and omitting these is likely to be a relatively small fraction of your error.

dangviethieu-hntv, 2018-08-03:
Thanks you. But I thinks why don't we use 1 boolean parameter, and set true when drone start action at waypoints, and set false when it finish, so we can calculate time drone fly and time drone doing action. That is my ideal, thank you very much !!!

hamishwillee, 2018-08-05:
I think the best you can do is monitor MISSION_ITEM_REACHED, which the autopilot must send whenever it gets to a new item. This is not timestamped, but it would give you a good approximation.

julianoes, 2024-05-27:
This should probably happen on the autopilot side and not the MAVSDK side.

closed 2024-05-27 by julianoes