mavlink/MAVSDK #478: Photo/Video saved location

damadamn, 2018-08-01:
HI, I would like to know where are stored photo/video after using command like `take_photo()`.
I'm working on an Intel Aero RTF, and I don't find any specific folders to store the photo, even after using this command

julianoes, 2018-08-02:
I suggest you check the Aero docs. Or you can ask @lbegani.

hamishwillee, 2018-08-02:
The information is supposed to be returned by [CAMERA_IMAGE_CAPTURED](https://mavlink.io/en/messages/common.html#CAMERA_IMAGE_CAPTURED).file_url

We should probably store that sort of information @julianoes . Problem though - I don't think there is any way to iterate through the server to get all historic images, so we should ask for that feature (or something to allow it.

lbegani, 2018-08-02:
@damadamn Depends on the handler (of the mavlink message) where it stores them locally.
If you are using `camera-manager` , by default the media is stored in `/tmp` directory. The path can be configured in conf file. Hosting the media files in server(http etc) and returning the URL is not yet supported by `camera-manager`

damadamn, 2018-08-07:
When I use the command line `gst-launch-1.0 v4l2src device=/dev/video13 num-buffers=1 ! jpegenc ! filesink location="rs.jpg"`, the picture is taken from the front camera ( a RealSense R200) and I find it inside the main repertory (`/home/root`), but that's a little ... dirty to use it inside a C/C++ program with `system()` function; that's why I would like to use `take_photo()`

julianoes, 2024-05-27:
Looks all very outdated and unrelated, to be honest.

closed 2024-05-27 by julianoes