mavlink/MAVSDK #481: Failed to Compile on Mac

mzahana, 2018-08-06:
Hi,

I tried to compile the library on Mac OS, El Capitan, using the `master` branch but failed.

Here is the output of `make default`:

```bash
-- The C compiler identification is AppleClang 7.3.0.7030031
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.dylib (found version "7.43.0")
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - failed
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- Found PythonInterp: /usr/bin/python (found version "2.7.10")
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   zlib

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/abdkadma/Desktop/src/DronecodeSDK/build/default
Build dir: build/default
[40/140] Building CXX object core/CMak...dronecore.dir/call_every_handler.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/call_every_handler.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DAPPLE -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -Ithird_party/zlib -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecore.dir/call_every_handler.cpp.o -MF core/CMakeFiles/dronecore.dir/call_every_handler.cpp.o.d -o core/CMakeFiles/dronecore.dir/call_every_handler.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/call_every_handler.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[41/140] Building CXX object core/CMakeFiles/dronecore.dir/connection.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/connection.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DAPPLE -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -Ithird_party/zlib -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecore.dir/connection.cpp.o -MF core/CMakeFiles/dronecore.dir/connection.cpp.o.d -o core/CMakeFiles/dronecore.dir/connection.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/connection.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[42/140] Building CXX object core/CMakeFiles/dronecore.dir/curl_wrapper.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/curl_wrapper.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DAPPLE -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -Ithird_party/zlib -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecore.dir/curl_wrapper.cpp.o -MF core/CMakeFiles/dronecore.dir/curl_wrapper.cpp.o.d -o core/CMakeFiles/dronecore.dir/curl_wrapper.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/curl_wrapper.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[43/140] Building CXX object core/CMakeFiles/dronecore.dir/system.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/system.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DAPPLE -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -Ithird_party/zlib -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecore.dir/system.cpp.o -MF core/CMakeFiles/dronecore.dir/system.cpp.o.d -o core/CMakeFiles/dronecore.dir/system.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/system.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[44/140] Building CXX object core/CMakeFiles/dronecore.dir/system_impl.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/system_impl.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DAPPLE -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -Ithird_party/zlib -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecore.dir/system_impl.cpp.o -MF core/CMakeFiles/dronecore.dir/system_impl.cpp.o.d -o core/CMakeFiles/dronecore.dir/system_impl.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/system_impl.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[45/140] Building CXX object core/CMakeFiles/dronecore.dir/dronecore.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/dronecore.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DAPPLE -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -Ithird_party/zlib -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecore.dir/dronecore.cpp.o -MF core/CMakeFiles/dronecore.dir/dronecore.cpp.o.d -o core/CMakeFiles/dronecore.dir/dronecore.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/dronecore.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[46/140] Building CXX object core/CMak...les/dronecore.dir/dronecore_impl.cpp.o
FAILED: core/CMakeFiles/dronecore.dir/dronecore_impl.cpp.o
[... cut ...] /dronecore.dir/global_include.cpp.o -MF core/CMakeFiles/dronecore.dir/global_include.cpp.o.d -o core/CMakeFiles/dronecore.dir/global_include.cpp.o -c /Users/abdkadma/Desktop/src/DronecodeSDK/core/global_include.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[49/140] Building CXX object third_par...MakeFiles/gtest.dir/src/gtest-all.cc.o
ninja: build stopped: subcommand failed.
make: *** [default] Error 1
```

julianoes, 2018-08-06:
It looks to me that you need to update your XCode toolchain, you are on `AppleClang 7.3.0.7030031`, and up to date is `AppleClang 9.1.0.9020039`.

If you can't update the toolchain, I suggest that you remove the `-Werror` flag and live with the warnings.

mzahana, 2018-08-07:
After updating Mac OS to Sierra and updating Xcode, I could compile the SDK.

closed 2018-08-07 by mzahana

julianoes, 2018-08-07:
Great, thanks for reporting back.

ayaanhaider, 2019-01-15:
@mzahana Hi! Can you suggest a way to upgrade Xcode? I am running Mac OS Sierra version 10.12.6 and Xcode version 9.2. To get the latest Xcode version I would need to upgrade to Mojave which I prefer not to. Thanks.

ayaanhaider, 2019-01-15:
Hi, 
I tried building it using Mac OS Sierra version 10.12.6 and Xcode version 9.2 and I get the following error:

```
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.dylib (found version "7.51.0") 
CMake Deprecation Warning at third_party/tinyxml2/CMakeLists.txt:11 (cmake_policy):
  The OLD behavior for policy CMP0063 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - failed
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Version: v0.12.0-dirty
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- Found PythonInterp: /Library/Frameworks/Python.framework/Versions/2.7/bin/python (found version "2.7.14") 
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   zlib

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/ayaanhaider/DronecodeSDK/build/default
Build dir: build/default
[33/160] Building CXX object core/CMakeFil...dronecode_sdk.dir/call_every_handler.cpp.
FAILED: core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -Ithird_party/zlib -Icore -I../../core/include -I../../core -isystem ../../third_party/tinyxml2 -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Werror -Wextra -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o -MF core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o.d -o core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o -c ../../core/call_every_handler.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[34/160] Building CXX object core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o
FAILED: core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -Ithird_party/zlib -Icore -I../../core/include -I../../core -isystem ../../third_party/tinyxml2 -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Werror -Wextra -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o -MF core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o.d -o core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o -c ../../core/connection.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[35/160] Building CXX object core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o
FAILED: core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -Ithird_party/zlib -Icore -I../../core/include -I../../core -isystem ../../third_party/tinyxml2 -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Werror -Wextra -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o -MF core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o.d -o core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o -c ../../core/curl_wrapper.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[38/160] Building CXX object third_party/c...c/src/CMakeFiles/cpp_rsc.dir/cpp_rsc.cpp.
ninja: build stopped: subcommand failed.
make: *** [default] Error 1

```
Any help would be much appreciated to solve this issue.

mzahana, 2019-01-16:
I am sorry, but I have not been a Mac user for a long time and I can't help
debugging this issue for now.