mavlink/MAVSDK #483: Failed to compile custom app on Mac

mzahana, 2018-08-07:
Hi,

I compiled the SDK on Mac OS (Sierra) and did system-wide install. However there is no `sudo ldconfig` (which is used on Linux to update the linker cache) in Mac. **What is the equivalent way to do it in Mac?**

I also followed the sdk guide (https://sdk.dronecode.org/en/guide/toolchain.html) in order to compile a custom C++ app, and created a custom app and the corresponding CMakeLists.txt.

However, when I run `make`, I get,
```bash
[ 50%] Building CXX object CMakeFiles/formation_gs.dir/formation_gs.cpp.o
/Users/abdkadma/Desktop/src/DroneCode_apps/formation_gs.cpp:15:10: fatal error:
      'dronecode_sdk/action.h' file not found
#include <dronecode_sdk/action.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/formation_gs.dir/formation_gs.cpp.o] Error 1
make[1]: *** [CMakeFiles/formation_gs.dir/all] Error 2
make: *** [all] Error 2
```

So, what could be the problem here? Is it because the linker cache is not updated?

mzahana, 2018-08-07:
Issue is solved by specifying the include and lib system directories in the CMakeLists.txt.

Would the tutorials need to be updated in order to account for this?

```bash
if(NOT MSVC)
    add_definitions("-std=c++11 -Wall -Wextra -Werror")
    include_directories(/usr/local/include)
    link_directories(/usr/local/lib)
else()
    add_definitions("-std=c++11 -WX -W2")
    include_directories(${CMAKE_SOURCE_DIR}/../../install/include)
    link_directories(${CMAKE_SOURCE_DIR}/../../install/lib)
endif()
```

mzahana, 2018-08-07:
Also, I would like to mention that I used the `develop` branch.

hamishwillee, 2018-08-07:
> Also, I would like to mention that I used the develop branch.

That is the correct branch to use. Develop is for ongoing development, master is for the last official release.

Regarding the install directories, did you install dronecode system wide or locally?  If you installed locally, then you do need to tell the system where the includes are: https://sdk.dronecode.org/en/guide/toolchain.html#sdk_local_install
My understanding is that if you installed system wide you shouldn't have to. 

IF you did and this is needed then yes, we will have to update a bunch of things, including test scripts and docs.

julianoes, 2018-08-14:
> However there is no sudo ldconfig (which is used on Linux to update the linker cache) in Mac. What is the equivalent way to do it in Mac?

https://stackoverflow.com/questions/1451047/ldconfig-equivalent-in-mac-os-x

julianoes, 2018-08-17:
Please re-open if this is still a problem.

closed 2018-08-17 by julianoes