mavlink/MAVSDK #486: Get a vector of systems

mzahana, 2018-08-11:
Hi.

I am working with multiple vehicles. [In the docs](https://sdk.dronecode.org/en/guide/connections.html), it's mentioned how to access a system by
```c++
DronecodeSDK dc;
//... 
//Use UUID named uuid to get the associated System
System &system = dc.system(uuid);
```

What is a correct way to create a variable number of multiple systems?

Thanks.

julianoes, 2018-08-13:
You should be able to get all system like described here:
https://sdk.dronecode.org/en/guide/connections.html#iterating-all-systems

However, for now you need to make sure yourself that all vehicles have different system IDs.

mzahana, 2018-08-13:
@julianoes so, I am looking into how to get `system_id` instead of uuid. I found this: https://github.com/Dronecode/DronecodeSDK/blob/develop/core/system_impl.cpp#L518

Would do that do the job?

but it is not exposed in [system.cpp](https://github.com/Dronecode/DronecodeSDK/blob/develop/core/system.cpp)

julianoes, 2018-08-14:
@mzahana is it not possible to use the UUIDs?

mzahana, 2018-08-14:
@julianoes Of course it is possible. It is not about possibility, it's about convenience.

Assume there is a set of 50 drones (maybe more!), which are labeled from 1 to 50. How you would you tell the user that drone 19 and 47 have some issues and require some checking?

I guess it's harder for the user to look for drones that have UUID 5345213345 and 6968987123 than just 19 and 47, no?

julianoes, 2018-08-14:
If something is automated in software UUID is more "convenient" because you it's unique.

I suggest you implement the method to get all system IDs yourself since you're already changed the API to work with system IDs instead of UUIDs.

mzahana, 2018-08-20:
@julianoes Would exposing [get_syste_id](https://github.com/Dronecode/DronecodeSDK/blob/develop/core/system_impl.cpp#L518) in [system.cpp](https://github.com/Dronecode/DronecodeSDK/blob/develop/core/system.cpp) do any harm?

julianoes, 2018-08-20:
> @julianoes Would exposing get_system_id in system.cpp do any harm?

I guess it doesn't :smile:, so we might as well add it.

julianoes, 2019-10-10:
Work in progress here: #694.

julianoes, 2020-11-20:
This has been added, see: https://mavsdk.mavlink.io/develop/en/cpp/api_changes.html#v0330

closed 2020-11-20 by julianoes