mavlink/MAVSDK #487: How to connect to a real vehicle?

zsyfight, 2018-08-13:
I want to run a example mission on a real vehicle , but I don't know how to set the serial connect protocol formation.

zsyfight, 2018-08-13:
Can I run the mission by this?
./fly_mission serial:///dev/ttyUSB0:57600

julianoes, 2018-08-13:
@zsyfight hi, how are you connecting to the vehicle? Wifi, 3DR radio, USB cable?

zsyfight, 2018-08-13:
@julianoes  Thanks for you response. I want to connecting the vehicle (3DR IRIS+) by the 3DR radio.What I want to do is to communicate with the vehicle by the radio, and upload the waypoints (just like the waypoints in the fly_mission.cpp) to the vehicle. so what should i do?

julianoes, 2018-08-13:
@zsyfight ok, if you're 3DR radio is connected to `/dev/ttyUSB0`, your line should work:
```
 ./fly_mission serial:///dev/ttyUSB0:57600
```

zsyfight, 2018-08-14:
@julianoes I just changed the lines as i have pasted, i get the result shown in the terminal : 
Waiting to discover system...
and without anymore information. So is there something should be configure in the fly_mission.cpp for the real vehicle connecting?

julianoes, 2018-08-14:
Does the connection over that radio work with QGC?

zsyfight, 2018-08-15:
I connected the vehicle with the radio, and then i open the QGC i get the error result:
~/DronecodeSDK/example/fly_mission/build$ ./fly_mission serial:///dev/ttyUSB0:57600
Waiting to discover system...
[10:20:48|Debug] New: System ID: 1 Comp ID: 1 (dronecode_sdk_impl.cpp:292)
[10:20:48|Debug] Component Autopilot added. (system_impl.cpp:339)
[10:20:53|Debug] Found 1 component(s). (system_impl.cpp:466)
[10:20:53|Debug] Discovered 3833204329145446450 (system_impl.cpp:468)
Discovered system with UUID: 3833204329145446450
Waiting for system to be ready
[10:20:53|Debug] New: System ID: 51 Comp ID: 68 (dronecode_sdk_impl.cpp:292)
[10:20:53|Debug] Component Unsupported component added. (system_impl.cpp:339)
[10:20:53|Debug] sysid: 51 (dronecode_sdk_impl.cpp:78)
[10:20:54|Error] Error: get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:440)
[10:20:54|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:242)
Waiting for system to be ready
[10:20:54|Error] Error: get param busy timeout: MPC_XY_CRUISE (mavlink_parameters.cpp:440)
[10:20:54|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:242)
[10:20:55|Error] Error: get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:440)
[10:20:55|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:242)
Waiting for system to be ready
[10:20:55|Error] Error: get param busy timeout: CAL_GYRO0_ID (mavlink_parameters.cpp:440)
[10:20:55|Error] Error: Param for gyro cal failed. (telemetry_impl.cpp:570)
[10:20:55|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:242)
[10:20:56|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:242)
[10:20:56|Error] Error: get param busy timeout: CAL_MAG0_ID (mavlink_parameters.cpp:440)
[10:20:56|Error] Error: Param for mag cal failed. (telemetry_impl.cpp:592)
Waiting for system to be ready
[10:20:56|Info ] heartbeats timed out (system_impl.cpp:282)
[10:20:56|Debug] Lost 3833204329145446450 (dronecode_sdk_impl.cpp:319)
System with UUID timed out: 3833204329145446450
Exiting.

zsyfight, 2018-08-24:
It's not a communication problem, the vehicle can connect with the QGC normally. Maybe it is a SDK problem.

julianoes, 2018-08-24:
What firmware version is running on the drone?

zsyfight, 2018-08-26:
The Firmware version is 1.8.0dev on the drone.

julianoes, 2018-08-27:
@zsyfight thanks, ok I will try to reproduce this.

zsyfight, 2018-09-28:
@julianoes how about the results of the problem? Did you get the connection  to the real vehicle?

julianoes, 2018-09-30:
@zsyfight please check again, this could be fixed with #565.

lalal321, 2018-10-08:
well, I get a problem when I connect a vehicle via a 3DR radio, and it shown as below: 

hello@hello:~/DronecodeSDK/example/fly_mission/build$ ./fly_mission serial:///dev/ttyUSB0:57600
Waiting to discover system...
[03:34:07|Debug] New: System ID: 1 Comp ID: 1 (dronecore_impl.cpp:292)
[03:34:07|Debug] Component Autopilot added. (system_impl.cpp:339)
[03:34:08|Debug] Found 1 component(s). (system_impl.cpp:466)
[03:34:08|Debug] Discovered 3545255410840907825 (system_impl.cpp:468)
Discovered system with UUID: 3545255410840907825
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready

It shows "waiting for system to be ready" consistently , I think it may be initializing the vehicle , but the vehicle is health in fact， it should to arm after the initialization, but it didn't. Do you know what's wrong with my vehicle? thank you. @julianoes

hamishwillee, 2018-10-08:
How do you *know* the vehicle is healthy?

I would modify the example to check the health / possible causes of the problem as shown in  https://sdk.dronecode.org/en/guide/taking_off_landing.html#health-check

julianoes, 2018-10-08:
Thanks @hamishwillee.

@lalal321 if the problem persists, please open a new issue. This issue is about an actual communication problem.

julianoes, 2018-10-08:
I'm closing this issue. @zsyfight if the problem persists, please re-open.

closed 2018-10-08 by julianoes

zsyfight, 2018-10-08:
oh,it's all right, and thank you very much for your help! @julianoes