mavlink/MAVSDK #494: Does this support multipe drones ?

Seunghwan17, 2018-08-16:
Who have to test multiple drones using dronecode sdk? In my case, if I tried to connect multiple drones, dronecodesdk doesn’t get some parameters. like accelometer.

I had experience and test with dronekit and QGC already. But, only dronecode sdk make some trouble that I say above.

If you have some experience plz share me some experience.

Seunghwan17, 2018-08-16:
I means the code show “get param beasy time out”.

julianoes, 2018-08-16:
Ok, I currently don't have a lot of bandwidth to debug this. So the more output you provide here, the better.

Also, please provide instructions how I can reproduce this. What are the commands the start the simulation, etc. Then I might be able to give this a try over the weekend.

Seunghwan17, 2018-08-16:
I make three dronecode system. And their telemetry is 3DR radio 433mhz. 

It works good if one drone make connection. 
But, if I make three drones connections  the code show me 3connections. Then it shows me some error “get param busy time out”......

Now I use some customized mavlink mode to reduce telemetry packet.

Seunghwan17, 2018-08-16:
By the way, why do dronecode try to get parameters 3 times? In my test, To get parameters make big traffic. Although I reduce mavlink message packet, but to get parameters doesn’t reduced, doesn’t it?

julianoes, 2018-08-16:
The action plugin caches some parameters for the speed setting, RTL altitude setting, etc. This is actually very little traffic because it's only the few parameters which are actually required instead of receiving all ~800 parameters.

julianoes, 2018-08-16:
Do you also test it in simulation? I believe there is support for multiple drones in SITL?

Jaeyoung-Lim, 2018-08-16:
@Seunghwan17 What is the configuration with your radio? You might want to directy connect the serial to the computer to eliminate the possibility of the radios being misconfigured

Seunghwan17, 2018-08-16:
@julianoes Really? I didn’t know that. you mean dronecode action plugin just few parameters. It is very strange to get patameters is fail.

I don’t test in SITL not yet.

@Jaeyoung-Lim What do configuration mean? I set 3paired 3DR radio has different net ID and 57600 baudrate.

Seunghwan17, 2018-08-16:
And I would try direct connect next day. Thanks.

julianoes, 2018-08-16:
Do you follow this? https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

Seunghwan17, 2018-08-17:
@julianoes @Jaeyoung-Lim Um... I would like to test SITL. I tested the directly connect to serial. The error message doesn't showed now. I think @Jaeyoung-Lim is right.  My radio has some misconfiguration. I couldn't understand now why my 433mhz radio is too late to get parameter. Maybe I change some configuration and more test with dronecode and radio.

closed 2018-08-17 by Seunghwan17