mavlink/MAVSDK #502: System Reboot

mzahana, 2018-08-20:
Hi.

In QGC, I can reboot the fight controller from the Parameters tab->Tools menu.

Is there a method in the SDK to reboot a system?

If not, I may give it a try in send a PR.

I guess [MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN](https://mavlink.io/en/messages/common.html#MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN) would be used, but have not checked yet.

Thanks.

hamishwillee, 2018-08-20:
I think that is the correct mavlink command, and no, I do not think that we have implemented it. I think it would be useful as an Action.

mrpollo, 2018-09-03:
For anyone reading and is interested in contributing, as @hamishwillee mentions this functionality would be great in the `Action` class, here's a link to the Action Implementation class, sending a message is a breeze with the SDK [`plugins/action/action_impl.cpp@ActionImpl:arm`](https://github.com/Dronecode/DronecodeSDK/blob/develop/plugins/action/action_impl.cpp#L52)

commit c266590e (2018-09-30): Fix #502. Add Action::reboot()
commit de61562d (2018-09-30): Merge pull request #551 from dbowerman/system_reboot

Fix #502. Add Action::reboot()