mavlink/MAVSDK #533: How to change PWM output signal?

maitham1, 2018-09-05:
I'm trying to change the PWM output using dronecode sdk. I don't need to run a mission or anything, I want to do a static test where I can simply change values. I seem to be running out of options in terms of simply changing pwm outputs within a script in a reliable manner (NSH pwm test works but only for main out). Is there anyway of currently doing this?

hamishwillee, 2018-09-05:
I don't think this is possible with the SDK because there is AFAIK no API to support it AND this won't necessarily be supported by default in PX4. 

Without having tried this  (i.e. maybe not correct) .... to make it work in PX4 you'd have to look at adding a mixer file to map control inputs to outputs - http://dev.px4.io/en/concept/mixing.html
To drive it you would need to send the appropriate MAVLink message (or have an onboard app that does the work for you).

hamishwillee, 2018-09-05:
PS The people who might help with this are possibly at Interdrone. 

@bkueng Can you comment about PX4 support? As I understand it the mixer is what defines what comes out of the MAIN/AUX outputs - for certain inputs (control group UORB topics?). Does the mapping between MAVLink and those UORB topics exist - ie if you create a mixer and send the appropriate MAVLink message will this work?

julianoes, 2018-09-10:
@maitham1 this will need some work for the SDK.

In the meantime, if `pwm test` works, you should try to make `pwm test` work for the aux output as well. Note that there is a `-d` argument which should enable you to choose the device, e.g. `/dev/pwm_output0` or `/dev/pwm_output1`.

julianoes, 2024-05-27:
There is now the set_actuator class:

https://github.com/mavlink/MAVSDK/blob/68c8041bc911e31f99c8575a042d6b3919583b39/src/mavsdk/plugins/action/include/plugins/action/action.h#L447-L456

closed 2024-05-27 by julianoes