mavlink/MAVSDK #534: Could Dronecode send a RTK message?

Seunghwan17, 2018-09-07:
I test RTK GPS now. The product is drotek's.
As I know, QGC send a RTK message. But I don't know droncode can do that.
I want to discuss about it. 
Thank you.

julianoes, 2018-09-10:
@Seunghwan17 ok sure let's discuss it. First I need some more detailed information about the GPS and then we need to figure out what exactly is needed from the SDK.

Seunghwan17, 2018-09-17:
Maybe, I think the SDK could detect base RTK module and send this message(https://mavlink.io/kr/messages/common.html#GPS_RTK) from base to the rover.
RTK module make RTCM3 data. The SDK need to pass a data use mavlink protocol... 
But now I don't know well about RTK and mavlink. I am sorry. 
Now I just tested with QGC.

julianoes, 2018-09-17:
We're open for contributions, e.g. a plugin that takes care of this.

AlexisTM, 2018-10-04:
For anybody willing to contribute on this matter, you will need a node reading the base_station and a plugin to forward the data to the Pixhawk.

An example of the base_station node using the GPSDriver (without using QT5) and publishing the RTCM messages (on ROS): 
https://github.com/AlexisTM/rtk_ros

An example for the plugin that needs to be implemented in Dronecore: 
https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/gps_rtk.cpp

julianoes, 2019-02-05:
Actually, this is a duplicate of #112. Closing this but I agree we should add it!

closed 2019-02-05 by julianoes