mavlink/MAVSDK #570: Gimbal plugin implementation idea

AlexisTM, 2018-10-04:
In the gimbal implementation, why isn't the roll implemented while it is available in the MAVLink message?

[develop@/plugins/gimbal/gimbal_impl.cpp](https://github.com/Dronecode/DronecodeSDK/blob/develop/plugins/gimbal/gimbal_impl.cpp)

Couldn't it be a better to use individual getter and setters for the gimbal, keeping in memory the command to apply?

```
void GimbalImpl::set_yaw(float yaw_deg)
{
   yaw = yaw_deg;
}
void GimbalImpl::set_pitch(float pitch_deg)
{
   pitch = pitch_deg;
}
void GimbalImpl::set_roll(float roll_deg)
{
   roll = roll_deg;
}
void GimbalImpl::set_pitch_yaw_roll(float yaw_deg, float pitch_deg, float roll_deg)
{
   roll = roll_deg;
   pitch = pitch_deg;
   yaw = yaw_deg;
}
```

And an apply function:

```
Gimbal::Result GimbalImpl::apply()
{
    MAVLinkCommands::CommandLong command{};

    command.command = MAV_CMD_DO_MOUNT_CONTROL;
    command.params.param1 = pitch;
    command.params.param2 = roll;
    command.params.param3 = yaw;
    command.params.param7 = float(MAV_MOUNT_MODE_MAVLINK_TARGETING);
    command.target_component_id = _parent->get_autopilot_id();

    return gimbal_result_from_command_result(_parent->send_command(command));
}
Gimbal::Result GimbalImpl::apply_async() { ... }
```

julianoes, 2018-10-04:
Do you have a use case where you need to set roll? I'm interested to hear about it, normally a gimbal just stabilizes roll.

AlexisTM, 2018-10-04:
@julianoes I don't have one personally, that's a very good point.

The feature I miss is to control the zoom (such as Peeper T10x) or shutter for other cameras.

AlexisTM, 2018-10-04:
On top of that, I am using the Offboard mode (PX4 1.8.0+)

julianoes, 2020-11-20:
No one else has asked for roll, so I'm closing this.

closed 2020-11-20 by julianoes