mavlink/MAVSDK #587: Enable serial and TCP connections for backend

julianoes, 2018-11-08:
The backend currently only supports UDP and the port is hardcoded for the backend_bin.

We should use `add_any_connection`, so that we can pass any connection URI.

JonasVautherin, 2018-11-12:
@shrit: Would you have examples illustrating what you mean (instead of the connection URI)?

julianoes, 2018-11-12:
We already support shortening where applicable, e.g. `udp://` will use the default port on localhost.

shrit, 2018-11-12:
well, actually after creating an instance of 
DronecodeSDK dc_;
we call the function 
dc_.add_any_connection(string);
the string variable contain the entire URI. For example, in the example of documentation we have:
udp://:14540
in the case twe are on udp on localhost  with the port 14540.

A rare use case, If we need to control several vehicles, several instance of controller should be created and we need to specify for each on the port and the connection type.
For this reason, I thought of simplifying even more for the users, like a user could type -P 14540 14550 14560, that is going to take a vector of port number for a set of users.

julianoes, 2018-11-12:
I would argue that's too rare of a use case. I would then just have the cli be a list, e.g.: `foo udp://:14540 udp://:14541` and then loop through it.

And note that you can have several vehicles connecting on the same UDP port at the same time, so you don't need a port per vehicle.

PR #730 (2019-04-28): Enable backend to accept connection URL
This allows the backend_bin to be started with any connection URL.

Resolves #587.
closed 2019-04-28 by JonasVautherin