mavlink/MAVSDK #603: Trouble installing backend

joe-marrone, 2018-11-21:
I ran into an issue trying to install the backend using instructions:

git clone https://github.com/Dronecode/DronecodeSDK --recursive
cd DronecodeSDK
make BUILD_BACKEND=1

Then when trying to run:

./build/default/backend/src/backend_bin

 I get the error message 

joe@joe-VirtualBox:~/DronecodeSDK$ /build/default/backend/src/backend_bin
bash: /build/default/backend/src/backend_bin: No such file or directory

I searched through all directories and did not find the file backend_bin.  

Any idea what went wrong?
Thanks
Joe

JonasVautherin, 2018-11-21:
Did the build actually work? Sounds like the build failed. 

On which system are you running?

joe-marrone, 2018-11-21:
@JonasVautherin , I am running the PX4 SITL on ubuntu 16.04.  I have the GUI running and can input individual commands OK.  What I want to do is install the DronecodeSDK-Python wrapper, so installing the backend is a prerequisite.  Evidently is did not install, but I don't know what went wrong.

Joe

JonasVautherin, 2018-11-21:
My guess would be that the gRPC build failed because you don't have go (go-lang) installed on your system. Can you check if go-lang is installed on your system? If not, install it and run the build again.

If it still fails, I will need to see the compiler output, because it probably says why it failed :-).

joe-marrone, 2018-11-21:
Is that the Go language?  If so how do I download it?
Sorry for the dumb questions.

Joe

JonasVautherin, 2018-11-21:
I am not on Ubuntu, but what about `apt install golang`?

hamishwillee, 2018-11-21:
That is worth trying, but there may be no official packet. I haven't done this, but see https://github.com/golang/go/wiki/Ubuntu for an alternative.

@JonasVautherin When this is resolve, we need a "Building the backend" section added here: https://sdk.dronecode.org/en/contributing/build.html for all platforms. Are there any other dependencies? What about how to do this on macOS and Windows?

JonasVautherin, 2018-11-21:
I believe `golang` exists on Ubuntu (tried in a docker just now :-)).

@hamishwillee: you're perfectly right, I'll do that when I find time! I have Linux and macOS, no idea about Windows, though ;-).

joe-marrone, 2018-11-21:
OK, this command worked and Go installed:
    sudo apt-get install golang

However when I ran 
    make BUILD_BACKEND=1

it gave this message, which was the same as before loading golang.
     Build dir: build/default
    ninja: no work to do.

So I don't think it installed the backend yet.

Another clue is that the directory **DronecodeSDK/build/default/backend**  does not exist, ie, the default folder does not contain a backend folder.

Joe

JonasVautherin, 2018-11-21:
Do you have something at all in `DronecodeSDK/build/default`?

Would you mind running a clean build, again, with:

```
make distclean
make BUILD_BACKEND=1
```

joe-marrone, 2018-11-21:
@JonasVautherin , the cleanup and re-make worked like a charm.  I now have all the dependencies for loading the Python wrapper.  BTW, that was a huge build, took almost a half hour.
Thanks for the very quick response.
Joe

JonasVautherin, 2018-11-21:
Yes, what's taking time is mainly cloning the dependencies actually :confused:. I'd like to work on improving that in the future.

closed 2018-11-21 by JonasVautherin

julianoes, 2018-11-22:
@JonasVautherin: is it added to the Python readme?

JonasVautherin, 2018-11-22:
Better: I improved the build so that the build fails fast ([see PR here](https://github.com/Dronecode/DronecodeSDK/pull/604)).

I wouldn't put instructions about building the backend in the Python readme.

julianoes, 2018-11-22:
> I wouldn't put instructions about building the backend in the Python readme.

But then they need to be linked. It's just that all info needs to be there, otherwise it's not useable.

hamishwillee, 2018-11-22:
I added build instructions you can link to here: https://github.com/dronecore/sdk_docs/pull/146 ... once you have reviewed/merged. Feel free to merge and iterate yourself if you need to create the link target.

joe-marrone, 2018-12-05:
Hello again, I have another issue.
when installing the Python wrapper everything appeared to install OK.  But when I tried to run the takeoff_and_landing script it threw this error:

joe@joe-VirtualBox:~/src/Firmware/DronecodeSDK-Python$ examples/takeoff_and_land.py
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 5, in <module>
    from dronecode_sdk import connect as dronecode_sdk_connect
ModuleNotFoundError: No module named 'dronecode_sdk'

What next?
Joe

hamishwillee, 2018-12-05:
@joe-marrone Raise a new issue in DronecodeSDK-Python - this issue is closed. Normally I'd say use the [support channels](https://sdk.dronecode.org/en/#getting-help) but this could well be a bug.