mavlink/MAVSDK #609: Spurious send tries in backend

julianoes, 2018-11-23:
When the backend is started, something tries to send messages even though we have not discovered a vehicle.

The output is this:
```
[03:39:09|Info ] DronecodeSDK version: 0.9.0 (dronecode_sdk_impl.cpp:24)
[03:39:09|Debug] New: System ID: 0 Comp ID: 0 (dronecode_sdk_impl.cpp:284)
[03:39:09|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:40)
[03:39:09|Info ] Server started (grpc_server.cpp:24)
[03:39:09|Info ] Waiting to discover system... (connection_initiator.h:57)
[03:39:10|Error] No known remotes (udp_connection.cpp:130)
[03:39:10|Error] send fail (dronecode_sdk_impl.cpp:90)
```

Steps to reproduce:
1. Don't run PX4 SITL, make sure nothing would connect.
2. `make BUILD_BACKEND=1 && build/default/backend/src/backend_bin`

jarednm, 2019-01-07:
Any update on this?  I have the same issue.

JonasVautherin, 2019-01-07:
As it is a "beginner" issue, we're keeping it open, hoping that somebody will feel like contributing :-)

snail914, 2019-02-15:
Bug found^^ My laptop is too slow to build the project, will send a pr tomorrow

hamishwillee, 2019-02-18:
You know that MAVLink components should publish heartbeats even if they haven't discovered other systems yet, right?

julianoes, 2019-02-19:
> You know that MAVLink components should publish heartbeats even if they haven't discovered other systems yet, right?

From my understanding that's not true. Systems such as drones should be sending heartbeats and ground stations are supposed to listen for heartbeats. Once they receive something, they can (don't have to) respond using heartbeats.

hamishwillee, 2019-02-20:
>> You know that MAVLink components should publish heartbeats even if they haven't discovered other systems yet, right?

> From my understanding that's not true. Systems such as drones should be sending heartbeats and ground stations are supposed to listen for heartbeats. Once they receive something, they can (don't have to) respond using heartbeats.

Not what the [spec says](https://mavlink.io/en/services/heartbeat.html#connectiondisconnection). Which does not mean you are wrong, but that we need this confirmed. 
I will confirm in dev call.

julianoes, 2019-02-20:
As far as I know, QGC does not do it either, at least on UDP because it would not even know where to send heartbeats. The port was not defined until recently when it was written in the port diagram. The other example is MAVROS which does send stuff by assuming the SITL port which just always happened to be the same (14557 I think).

hamishwillee, 2019-02-20:
As discussed in devcall. You have a good point, with the result that you now have action to work out best way to handle connection, vehicle discovery and system id allocation mechanisms. That'll learn ya!

julianoes, 2019-04-30:
Fixed by #674.

closed 2019-04-30 by julianoes