mavlink/MAVSDK #610: Remove hard-coded `-j8` argument for `make`

julianoes, 2018-11-26:
The current Makefile is sort of a wrapper script around cmake and that's a bit confusing.

Especially annoying is actually the hard-coded `-j8` argument for make. It would be better if this was taken from the CPU count or your settings for make (`MAKEFLAGS`).

https://github.com/Dronecode/DronecodeSDK/blob/ef45e4ddb7a8c5dcd0ac20df9a330afdaa9a7be6/Makefile#L22

More thoughts: https://stackoverflow.com/questions/10567890/parallel-make-set-j8-as-the-default-option

julianoes, 2018-12-02:
Hi @simrankedia,

1. The Makefile is used on Linux and macOS. For Windows we directly use cmake and don't use this Makefile helper.
2. I usually build with n if the CPU has n cores/hyperthreads.

simrankedia, 2018-12-03:
Hi @julianoes,
Thanks for the prompt reply. Just a clarification regarding point 2 above: 
The machine is supposed to be doing nothing else while in the build is going on? (trying to understand if we should use less than n cores for the build.)

julianoes, 2018-12-04:
> The machine is supposed to be doing nothing else while the build is going on? (trying to understand if we should use less than n cores for the build.)

You can still do other things but they will be slowed down. I think it's common practice to use `-jX` for `X` hypercores.

Check the build instructions here:
https://sdk.dronecode.org/en/contributing/build.html

JonasVautherin, 2018-12-04:
> You can still do other things but they will be slowed down. I think it's common practice to use -jX for X hypercores.

Agreed. And there is always `nice` that does process scheduling if you don't want it to take too many resources.

simrankedia, 2018-12-09:
Hi @julianoes 
I am done with the changes. However, looks like I don't have the permission to create a branch. Please guide me on how to submit the changes for review. 

Thanks.

julianoes, 2018-12-09:
Nice @simrankedia! So the way it works is that you fork this repository and then create a pull request from there.
You can find some instructions here: https://akrabat.com/the-beginners-guide-to-contributing-to-a-github-project/

simrankedia, 2018-12-09:
Hi @julianoes I have raised a pull request. Kindly have a look.

JonasVautherin, 2018-12-09:
@simrankedia: I see your branch [here](https://github.com/simrankedia/DronecodeSDK/tree/Issue610) on your fork, but not a PR. Are you sure you opened a pull request to this repo? From [your branch](https://github.com/simrankedia/DronecodeSDK/tree/Issue610), click "Pull Request", add a description (add "Resolves #610" in the description), and hit "Create pull request".

Then it will appear [here on the DronecodeSDK repo](https://github.com/Dronecode/DronecodeSDK/pulls) :-).

simrankedia, 2018-12-10:
Hi @JonasVautherin, Sorry my bad. I have raised the PR now.

PR #622 (2018-12-11): Resolves #610
Resolves #610
commit e802dcc5 (2018-12-11): Merge pull request #622 from simrankedia/Issue610

Resolves #610
closed 2018-12-11 by julianoes