mavlink/MAVSDK #611: Remove `-Werror` except for CI

julianoes, 2018-11-26:
We should probably disable to error on warnings (`-Werror`) except for CI runs.

Currently, `-Werror` is included here:
https://github.com/Dronecode/DronecodeSDK/blob/ef45e4ddb7a8c5dcd0ac20df9a330afdaa9a7be6/cmake/compiler_flags.cmake#L13

but it would be better if this was not set by default and added in CI.

JonasVautherin, 2018-11-26:
Why is that? If warnings are errors at the CI level, they are effectively errors at all levels. I don't get why we would disable it. For development? :thinking:

julianoes, 2018-11-26:
Because you don't want to error just because someone is using an odd or newer version of a compiler.
E.g. when someone is installing via brew or from AUR you generally don't have `-Werror` because it would break too many times.

PR #887 (2020-09-25): cmake: remove -Werror
I think it's a better idea to only use -Werror in CI because:
- This way the build is more robust when built for unknown environments with unknown compilers and compiler versions.
- Many developers get confused by warnings and are blocked assuming they are actual errors.

Fixes #611.
closed 2020-09-25 by julianoes