mavlink/MAVSDK #612: Cross-compile and deploy for Raspberry Pi

julianoes, 2018-11-26:
It would actually be very nice if we could also generate a .deb for Raspbian, so it can be installed easily on Raspberry Pi.

Building on Raspberry Pi is really quite slow, especially when you would add the gRPC backend.

JonasVautherin, 2018-11-26:
I'm not sure this is a beginner issue :sweat_smile:.

This means that you need to cross-compile all the libs for rpi, and last time I tried it was already not working for curl (I contributed to fix the bug, and I believe it was working in the end).

Still with curl, one issue we have is that we don't really compile it from sources but have this  "curl-android-ios" for Android and iOS, which is already some kind of workaround.

hamishwillee, 2018-11-26:
Love the idea.

More generally we still need an idiot proof deployment of all our runtimes for our supported platforms - ie no more pointing people to the build page at all.

julianoes, 2018-11-26:
It's good for an SDK beginner knowing cross-compilation :smile: 

> More generally we still need an idiot proof deployment of all our runtimes for our supported platforms - ie no more pointing people to the build page at all.

I agree.

julianoes, 2019-04-30:
This will be easier after #724.

ameya7295, 2019-06-05:
I'm taking this task.

JonasVautherin, 2019-06-05:
Wait, I can help with the latest `develop`! Which architecture do you need to build? `armv7`?

ameya7295, 2019-06-07:
yes, I would be building on armv7.

JonasVautherin, 2019-06-07:
Would you mind trying with dockcross? I think I successfully built the C++ SDK (I did not try the backend):

```sh
cd /path/to/DronecodeSDK
docker run --rm dockcross/linux-armv7 > ./dockcross
chmod +x dockcross
./dockcross cmake -DCMAKE_INSTALL_PREFIX=install_armv7 -Bbuild/armv7 -S.
./dockcross cmake --build build/armv7 --target install
```

And then I checked that a library was indeed an ARM binary:

```sh
file install_armv7/lib/libdronecode_sdk.so
install_armv7/lib/libdronecode_sdk.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, with debug_info, not stripped
```

JonasVautherin, 2019-06-07:
Seems like building the backend has issues, though. @ameya7295 if you feel like debugging that, that would be awesome. It seems like `FindZLIB.cmake` doesn't find zlib when building gRPC. However, it also says:

```
CMake variable ZLIB_ROOT is set to:

/work/build/armv7/third_party/install
```

and zlib is indeed installed there. But still it errors with:

```
CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
```

For debugging, I would either run the dockcross image interactively (`docker run --rm -it dockcross/linux-armv7 /bin/bash`) or try to build on a RPi natively (i.e. without dockcross), to see if it is reproducible there.

ameya7295, 2019-06-08:
okay, I would give this a try. thanks for sharing the required info.

JonasVautherin, 2019-06-08:
Now wondering if it could be related to [this cmake bug](https://gitlab.kitware.com/cmake/cmake/issues/19282)... :thinking:

julianoes, 2019-10-23:
Thanks to @JonasVautherin's [fix against Dockcross](https://github.com/dockcross/dockcross/pull/337) this should work now:

```
docker run --rm dockcross/linux-armv7 > ./dockcross-linux-armv7
chmod +x ./dockcross-linux-armv7
./dockcross-linux-armv7 cmake -DBUILD_BACKEND=ON -DBUILD_SHARED_LIBS=OFF -Bbuild/linux-armv7 -S.
./dockcross-linux-armv7 cmake --build build/linux-armv7 -j 8
```

Please re-open the issue if this is still a problem.

closed 2019-10-23 by julianoes