mavlink/MAVSDK #614: Arming fails without GPS but external vision position estimates

julianoes, 2018-11-27:
Apparently, we can't arm without GPS even if PX4 receives position estimates using vision.

This is something worth investigating.

jbeyerstedt, 2019-10-11:
This also affects rovers, or all vehicles, that might not need a (global) position at all to drive/ fly. So please also keep this use case in mind.

The FMU on my rover arms just fine using `commander arm` or QGC, but not using the Python MAVSDK and the `offboard_attitude.py` example:
```
-- Arming
Traceback (most recent call last):
  File "./main.py", line 77, in <module>
    loop.run_until_complete(run())
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "./main.py", line 40, in run
    await drone.action.arm()
  File "/Users/jannik/Development/MAVSDK-Python/mavsdk/generated/action.py", line 230, in arm
    raise ActionError(result, "arm()")
mavsdk.generated.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()
```

JonasVautherin, 2019-10-11:
@jbeyerstedt: Would you have a suggestion on how to fix that in MAVSDK? I personally have no clue :laughing:.

jbeyerstedt, 2019-10-14:
May I ask, why the `arm()` command changes the flight mode first? I don't know, why this might be useful, but I'm also not _that_ experienced with PX4.

As far as I have thought of it, I don't expect the `arm()` action to change the flight mode. I have understood the (C++) API Reference, that I need to arm first and then takeoff. So it's the same, as it is done with QGC. And the takeoff will lead to a position hold anyway.

Otherwise, the API must differentiate between different airframe types. I have to lookup, if the MAVSDK can know this. But that solution would be a pain to maintain and is not a "nice" design.

julianoes, 2019-10-14:
My original idea when implementing this was to be safe. By checking if the vehicle is landed and setting it to hold mode we can be sure that it won't takeoff (or otherwise shoot in the air) when it is armed.
One such a dangerous case would be if the vehicle is in a manual mode like stabilized with a spring loaded throttle stick at 50 %. If you arm the vehicle like that it will shoot up in the sky, of course depending on how much throttle it needs to takeoff.

However, I can see now that this is causing a lot of headaches and we need re-think it.

jbeyerstedt, 2019-10-14:
> One such a dangerous case would be if the vehicle is in a manual mode like stabilized with a spring loaded throttle stick at 50 %. If you arm the vehicle like that it will shoot up in the sky, of course depending on how much throttle it needs to takeoff.

Isn't arming prevented, when the throttle stick of not all the way down? At least that was the cat, as I tried to steer my rover FMU with QGC's on-screen controls and also using the quadcopter with a remote control. So normally this shouldn't be an issue.

julianoes, 2019-10-14:
Good point, that's right. I guess I'm still scared.

However, when you're in mission mode and you arm, it will actually take off and start flying.

jbeyerstedt, 2019-10-14:
You're right, that safety is important. But the user also has some responsibility. So there is always the question, how much safety checks the SDK should do and what is left to the user/developer, that is using the SDK. I don't know, if there's already a consensus about this in the MAVSDK project.

One solution might be to provide two methods, one just arming the drone and one with additional safety checks (or the current implementation plus a version without the additional step). But this might also lead to users, that blindly rely on the MAVSDK to do safety checks.

In any case, the documentation should be updated to clearly state, what the method actually does (after this discussion has come to a conclusion).

shrit, 2019-10-14:
I have been waiting for this feature for a long time, it is nice to find a consensus of ideas about this problem.
Actually, I remember being able to take off a quadcopter with a position estimator such as an optical flow camera in Gazebo, but this was without using the SDK.

julianoes, 2019-10-14:
We could work around the one case where we're in MISSION mode by checking if we're in MISSION mode and if so we switch to HOLD. For anything else, we just arm.

JonasVautherin, 2019-10-14:
Just to make sure I understand: PX4 now prevents you to arm under conditions that will make it jump in the air, but allows you to arm while in mission mode. "Just arming" for any mode that is not MISSION should therefore be safe, but relies solely on the autopilot.

Is that correct? If yes, then I'd be fine with @julianoes suggested workaround. It may also be worth adding a test for PX4 to make sure that it doesn't "shoot up in the sky" under those conditions =).

jbeyerstedt, 2019-10-14:
@JonasVautherin you're right (at least as far as I know). I haven't used the MISSION mode with a real drone yet (just SITL simulation).
But starting a mission, at least from QGC, will automatically start the flight and therefore arm the drone anyway. So maybe the case you want to prevent doesn't exist in reality? Or am I wrong?

JonasVautherin, 2019-10-14:
> But starting a mission, at least from QGC, will automatically start the flight and therefore arm the drone anyway. So maybe the case you want to prevent doesn't exist in reality? Or am I wrong?

I think that the "dangerous" scenario is the other way round. In QGC, if you "swipe" to start the mission, it will go in mission mode and arm. But say you are in MISSION mode, and just "arm" the vehicle, then it will takeoff and fly the mission. In the first case, that's clearly what you wanted (you swiped to start the mission). In the second case, you clearly armed, but it's not clear whether you were aware that the mission would start or not.

julianoes, 2019-10-15:
> PX4 now prevents you to arm under conditions that will make it jump in the air, but allows you to arm while in mission mode.

That's only my assumption. We have to verify it.

> In the second case, you clearly armed, but it's not clear whether you were aware that the mission would start or not.

Exactly!

yingshaoxo, 2019-10-24:
Why you guys talking a lot, just use a parameter `arm(force=False)` and it'll solve all sorts of problems.

Now, without that parameter, I can't even arm the drone (let motor run) in `Offboard Mode`.

___

To be clear, I connected a drone with a raspberry_pi, then use the raspberry_pi as a companion computer.

If I run the example `https://github.com/mavlink/MAVSDK-Python/blob/master/examples/offboard_position_ned.py`, codes running (but `arm function` will case `mavsdk.generated.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()` error). The real drone motor won't start.

Only if I use a `radio remote controller` to `arm` it first, then start `offboard_position_ned.py` program, it runs.

closed 2019-10-24 by julianoes

jbeyerstedt, 2019-10-24:
I just did it as you said. Arming works just fine now (at least with MAVSDK-Python).

The request about changing MAVSDK-Python is for the newly added methods, right? Because I didn't see an additional "arm safely" command. (Which is not needed, at least from my side.)

JonasVautherin, 2019-10-24:
> The request about changing MAVSDK-Python is for the newly added methods, right? 

Not sure I get that. So building `mavsdk_server` from sources will give you the latest bugfixes/improvements (here, we removed some checks in the arm command). For new features, you need to generate the python files as well.

> Because I didn't see an additional "arm safely" command.

You're right, we just "fixed" the arm command :+1:.

jbeyerstedt, 2019-10-24:
@JonasVautherin I was just curious, why yingshaoxo requested adapting MAVSDK-Python as well. Because no change is needed for the "bugfix" to work. But then I saw, that the pull request also introduced some new methods for the flight mode.

julianoes, 2019-10-25:
> But then I saw, that the pull request also introduced some new methods for the flight mode.

I don't think anything changed in the plugin API. Which code/diff do you mean?

jbeyerstedt, 2019-10-25:
I took a close look and in the end it's only one additional public method in src/core/system_impl.h, `SystemImpl::get_flight_mode()`:
https://github.com/mavlink/MAVSDK/pull/886/files?file-filters%5B%5D=.h#diff-e9e6ac77a8109ccb9b49b42b495b9f8aR131-R132

julianoes, 2019-10-25:
Right but that's only "public" internally. `SystemImpl` is not a public interface of the MAVSDK library, it's an interface for plugins to use.

yingshaoxo, 2019-10-29:
> I'll do that ASAP. In the meantime, what you can do:
> 
> 1. Build `mavsdk_server` from source (`develop`, so that it has your changes)
> 2. Run `mavsdk_server` manually on your machine
> 3. Run MAVSDK-Python against that `mavsdk_server`, with:
> 
> ```python
> from mavsdk import System
> 
> drone = System(mavsdk_server_address="localhost")
> await drone.connect()
> ```
> 
> It works like this: if you do `drone = System()`, it's the same as `drone = System(mavsdk_server_address=None)`, which tells MAVDSK-Python to run `mavsdk_server` for you in the background.
> 
> If you do `drone = System(mavsdk_server_address="localhost")` (or the IP of the machine running `mavsdk_server`, if it's not localhost), then MAVSDK-Python will just "connect" to it, without running `mavsdk_server`.
> 
> Hope it helps!

After I recompiled the backend_server, I run my script, I get:
```
Waiting for mavsdk_server to be ready...
Got MAVLink msg: COMMAND_ACK {command : 520, result : 0}
Got MAVLink msg: COMMAND_ACK {command : 511, result : 0}
Connected to mavsdk_server!
Got MAVLink msg: COMMAND_ACK {command : 520, result : 0}
-- Arming
Flight mode: LAND
COMMAND_DENIED_LANDED_STATE_UNKNOWN: 'Command denied, landed state is unknown'; origin: arm(); params: ()
-- Setting initial setpoint
-- Starting offboard
Got MAVLink msg: COMMAND_ACK {command : 176, result : 0}
OFFBOARD> Mode OFFBOARD
Flight mode: OFFBOARD
-- Go up 0.5 m
-- Go 1m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Back to start point: Go 0m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Go -1m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Back to start point: Go 0m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
Not in the air now.
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
Task exception was never retrieved
future: <Task finished coro=<print_flight_mode() done, defined at test.py:94> exception=<_Rendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>>
Traceback (most recent call last):
  File "test.py", line 99, in print_flight_mode
    async for flight_mode in drone.telemetry.flight_mode():
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.7/dist-packages/aiogrpc/utils.py", line 126, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.7/dist-packages/grpc/_channel.py", line 364, in __next__
    return self._next()
  File "/usr/local/lib/python3.7/dist-packages/grpc/_channel.py", line 358, in _next
    raise self
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>
Task exception was never retrieved
future: <Task finished coro=<print_altitude() done, defined at test.py:82> exception=<_Rendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>>
Traceback (most recent call last):
[... 139 lines cut ...]
            # await drone.action.kill()
            await drone.action.disarm()
            await asyncio.get_event_loop().shutdown_asyncgens()
            return

        #if is_in_air:
        #    was_in_air = is_in_air

        #if was_in_air and not is_in_air:
        #    print("Not in the air now.")
        #    # await drone.action.kill()
        #    await drone.action.disarm()
        #    await asyncio.get_event_loop().shutdown_asyncgens()
        #    return

if __name__ == "__main__":
    asyncio.get_event_loop().run_until_complete(run())
```

___

The proxy command I used for connecting the drone is :
```
mavproxy.py --master=/dev/ttyUSB0 --out=udp:127.0.0.1:14540 --out=udp:127.0.0.1:14550 --out=udp:127.0.0.1:14560 --out=udp:192.168.43.31:14540 --out=udp:192.168.43.31:14550 --out=udp:192.168.43.31:14560 --daemon &
```

julianoes, 2019-10-29:
The last version is probably not in the Python package yet.

JonasVautherin, 2019-10-29:
> My script is:

@yingshaoxo: Your script seems to not be doing what I'm saying in the part you quoted. Is that right? I said:

> 1. Build mavsdk_server from source (develop, so that it has your changes)
> 2. Run mavsdk_server manually on your machine
> 3. Run MAVSDK-Python against that mavsdk_server, with:
> 
> ```python
> from mavsdk import System
> 
> drone = System(mavsdk_server_address="localhost")
> await drone.connect()
> ```

And your script does:

```python
drone = System()
await drone.connect()
```

Can you try to follow my steps above? You will need all 3 of them, otherwise it won't be running against the `mavsdk_server` binary that you compile with the latest changes.

yingshaoxo, 2019-10-29:
This is what I did:

1. update
```
cd MAVSDK
git fetch --all
git reset --hard origin/master
```

2. compile
```
cmake -DBUILD_BACKEND=ON -Bbuild/default -H.
cmake --build build/default
```

3. go to `MAVSDK/build/default/src/backend/src`, run:
```
./mavsdk_server &
```

4. run mavproxy:
```
mavproxy.py --master=/dev/ttyUSB0 --out=udp:127.0.0.1:14540 --out=udp:127.0.0.1:14550 --out=udp:127.0.0.1:14560 --out=udp:192.168.43.31:14540 --out=udp:192.168.43.31:14550 --out=udp:192.168.43.31:14560 --daemon &
```

5. run python script
```python
#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk import (OffboardError, PositionNedYaw)

async def run():
    # Init the drone
    drone = System(mavsdk_server_address='localhost')
    await drone.connect()

    # Set parameters
    await drone.param.set_float_param("MIS_TAKEOFF_ALT", 1.0)  # set takeoff height to 1 meter
    await drone.param.set_int_param("COM_TAKEOFF_ACT", 0)  # hold after takeoff
    await drone.param.set_int_param("COM_OBL_ACT", 0)  # 0: land if lost offboard signal; 1: hold if lost offboard signal

    # Start parallel tasks
    asyncio.ensure_future(print_altitude(drone))
    asyncio.ensure_future(print_flight_mode(drone))
    termination_task = asyncio.ensure_future(observe_is_in_air(drone))

    # Execute the maneuvers
    print("-- Arming")
    try:
        await drone.action.arm()
    except Exception as e:
        print(e)

    #print("-- Taking off")
    # await drone.action.takeoff()

    # await asyncio.sleep(10)  # stay in there for 10 seconds

    print("-- Setting initial setpoint")
    await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))

    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
[... 93 lines cut ...]
COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()
-- Setting initial setpoint
-- Starting offboard
Got MAVLink msg: COMMAND_ACK {command : 176, result : 0}
OFFBOARD> Mode OFFBOARD
Flight mode: OFFBOARD
-- Go up 0.5 m
-- Go 1m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Back to start point: Go 0m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Go -1m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Back to start point: Go 0m North, 0m East, -0.5m Down (actually 0.5m high from ground) within local coordinate system, turn to face East
-- Stopping offboard
Got MAVLink msg: COMMAND_ACK {command : 176, result : 1}
[10:45:40|Warn ] command temporarily rejected (176). (mavlink_commands.cpp:170)
Stopping offboard mode failed with error code: COMMAND_DENIED
-- Landing
Got MAVLink msg: COMMAND_ACK {command : 21, result : 0}
LAND> Mode LAND
Flight mode: LAND
APM: Landing at current position
[10:45:40|Debug] MAVLink: info: Landing at current position (system_impl.cpp:304)
Not in the air now.
```

7. but the drone still not in arm

JonasVautherin, 2019-10-29:
I think that's a different error message now:

```
-- Arming
Got MAVLink msg: COMMAND_ACK {command : 176, result : 1}
[10:45:34|Warn ] command temporarily rejected (176). (mavlink_commands.cpp:170)
COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()
```

`COMMAND_DENIED` means that it was denied by the autopilot, not by MAVSDK. Before, you had `COMMAND_DENIED_LANDED_STATE_UNKNOWN`, which was a check from MAVSDK. So it seems like it is fixed, as far as this issue is concerned :+1:. Or am I missing something?

> APM: Landing at current position

Does that mean you are running APM?

yingshaoxo, 2019-10-29:
> concerned 👍. Or am I missing something?

I'm running PX4

JonasVautherin, 2019-10-29:
Alright. Seems to me like your drone is not ready and returns `COMMAND_DENIED` when you try to arm. I would suggest you open a new issue describing your setup and showing the code that fails, so that we can start from there :blush:.

julianoes, 2019-10-29:
> git reset --hard origin/master

You need the develop branch.

PR #886 (2019-10-24): Remove switching to HOLD mode before arm and takeoff
This removes most of the workaround where we switch to HOLD before arming. This was mostly out of precaution but as discussed in #50 it can hopefully safely get dropped now.

This also includes some refactoring and cleanup of the action plugin.

For reviewing, please check the individual commits and messages.

Fixes #50, fixes #614, fixes #855.