mavlink/MAVSDK #616: Expose full mission list

sanderux, 2018-11-30:
We need access to the full mission list.
for this we would like to get a signal when the mission changed (finished uploading) and a way to read all items (not just then ones known to DC)

@julianoes

JonasVautherin, 2018-12-03:
> get a signal when the mission changed (finished uploading)

Is that in MAVLink? Say if I connect two instances from QGC and upload a mission from one, the other currently doesn't know about that, right?

> a way to read all items (not just then ones known to DC)

I guess that would be a new "mission" plugin, as it probably doesn't fit in the current one (or does it?)

hamishwillee, 2018-12-03:
> get a signal when the mission changed (finished uploading)

You can get completion from the SDK `upload_mission_async()` but there is no MAVLink message to notify of a full mission upload: https://mavlink.io/en/services/mission.html

> I guess that would be a new "mission" plugin, as it probably doesn't fit in the current one (or does it?)

It doesn't fit into the current plugin model of a "simple" API. Options are to add support for all types of mission item, to add an API to set or query the underlying mission params (ie allow a generic option for unknown params). I like the flexibility, but I bet @julianoes won't :-)

julianoes, 2018-12-04:
> I guess that would be a new "mission" plugin, as it probably doesn't fit in the current one (or does it?)

Yes I'd rather have some duplication than one complicated one with all the options that is just as hard as mavlink itself.

JonasVautherin, 2018-12-04:
> IMO a reasonable compromise option might be to add a mission API to download/upload to file or text buffer using the standard plaintext format. Then anyone who wanted to be "clever" could manually edit and round-trip the file.

My concern is that if we go down this road, we will quickly end up allowing "plaintext mavlink" a bit everywhere, and that's going against the SDK. We can try say "well, you can use that feature, but we may break it and we won't support you in using it", but people will probably still complain when it breaks, and they will still expect support for that.

I don't know, just a concern :-).

hamishwillee, 2018-12-05:
I don't think this breaks the SDK, because the plaintext mission format is defacto standard for mission file storage/retrieval. Entirely up to you guys though. The alternatives are to my mind not as clean.

JonasVautherin, 2018-12-05:
I see. Well that could be a possibility. @julianoes do you have an opinion on that?

julianoes, 2018-12-05:
Yes, we could have a small plugin just to convert these mission format files.

JonasVautherin, 2018-12-05:
Wouldn't that fit in the current mission plugin?

julianoes, 2018-12-05:
> Wouldn't that fit in the current mission plugin?

Not really because the mission items don't support all the settings and commands so then part of the file that you want to import is not supported which is actually the case right now as we already support the mission plan file.

sanderux, 2018-12-06:
> Is that in MAVLink? Say if I connect two instances from QGC and upload a mission from one, the other currently doesn't know about that, right?

We now listen for MISSION_ACK to know if the mission was updated

julianoes, 2019-02-05:
Done in #644.

closed 2019-02-05 by julianoes