mavlink/MAVSDK #640: Make default giving error

ayaanhaider, 2019-01-15:
Hi,
I am using Mac OS Sierra version 10.12.6 and Xcode version 9.2. After following the instruction: `make default` , I get the following error:
```

-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.dylib (found version "7.51.0") 
CMake Deprecation Warning at third_party/tinyxml2/CMakeLists.txt:11 (cmake_policy):
  The OLD behavior for policy CMP0063 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - failed
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Version: v0.12.0-dirty
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- Found PythonInterp: /Library/Frameworks/Python.framework/Versions/2.7/bin/python (found version "2.7.14") 
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   zlib

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/ayaanhaider/DronecodeSDK/build/default
Build dir: build/default
[33/160] Building CXX object core/CMakeFil...dronecode_sdk.dir/call_every_handler.cpp.
FAILED: core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -Ithird_party/zlib -Icore -I../../core/include -I../../core -isystem ../../third_party/tinyxml2 -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Werror -Wextra -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o -MF core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o.d -o core/CMakeFiles/dronecode_sdk.dir/call_every_handler.cpp.o -c ../../core/call_every_handler.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[34/160] Building CXX object core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o
FAILED: core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -Ithird_party/zlib -Icore -I../../core/include -I../../core -isystem ../../third_party/tinyxml2 -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Werror -Wextra -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o -MF core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o.d -o core/CMakeFiles/dronecode_sdk.dir/connection.cpp.o -c ../../core/connection.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[35/160] Building CXX object core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o
FAILED: core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -Ithird_party/zlib -Icore -I../../core/include -I../../core -isystem ../../third_party/tinyxml2 -isystem ../../third_party/mavlink/include -g -DDEBUG   -fno-exceptions -Wall -Werror -Wextra -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-unused-lambda-capture -Wno-old-style-cast -std=gnu++11 -MD -MT core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o -MF core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o.d -o core/CMakeFiles/dronecode_sdk.dir/curl_wrapper.cpp.o -c ../../core/curl_wrapper.cpp
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
[38/160] Building CXX object third_party/c...c/src/CMakeFiles/cpp_rsc.dir/cpp_rsc.cpp.
ninja: build stopped: subcommand failed.
make: *** [default] Error 1

```
Any help would be much appreciated to solve this issue.

JonasVautherin, 2019-01-16:
Which version of clang do you have? 

    $ clang --version

ayaanhaider, 2019-01-16:
It gave me the following clang version @JonasVautherin : 
```
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

JonasVautherin, 2019-01-16:
It actually looks like #481. You could try either to update clang, or to remove `-Wno-unused-lambda-capture` [here](https://github.com/Dronecode/DronecodeSDK/blob/develop/cmake/compiler_flags.cmake#L28-L31).

ayaanhaider, 2019-01-16:
I removed the line you told me to but still no luck and got a similar error message. I tried it on another computer which is running Mac OS Mojave and clang version 10. I got this error while running it on the second MacBook: 
```
-- The C compiler identification is AppleClang 10.0.0.10001044
-- The CXX compiler identification is AppleClang 10.0.0.10001044
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.dylib (found version "7.54.0") 
CMake Deprecation Warning at third_party/tinyxml2/CMakeLists.txt:11 (cmake_policy):
  The OLD behavior for policy CMP0063 will be removed from a future version
  of CMake.
  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - failed
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Version: v0.12.0-dirty
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.15") 
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
  MACOSX_RPATH is not specified for the following targets:
   zlib
This warning is for project developers.  Use -Wno-dev to suppress it.
-- Generating done
-- Build files have been written to: /Users/Shayaan/DronecodeSDK/build/default
Build dir: build/default
[1/160] Building C object third_party/...ib/CMakeFiles/zlibstatic.dir/gzwrite.o
FAILED: third_party/zlib/CMakeFiles/zlibstatic.dir/gzwrite.o 
/Library/Developer/CommandLineTools/usr/bin/cc -DAPPLE -I../../third_party/zlib -Ithird_party/zlib -I../../ -isystem ../../third_party/tinyxml2 -g -fPIC   -fno-exceptions -MD -MT third_party/zlib/CMakeFiles/zlibstatic.dir/gzwrite.o -MF third_party/zlib/CMakeFiles/zlibstatic.dir/gzwrite.o.d -o third_party/zlib/CMakeFiles/zlibstatic.dir/gzwrite.o   -c ../../third_party/zlib/gzwrite.c
error: unable to open output file 'third_party/zlib/CMakeFiles/zlibstatic.dir/gzwrite.o': 'Operation not permitted'
1 error generated.
[2/160] Building C object third_party/...ib/CMakeFiles/zlibstatic.dir/adler32.o
FAILED: third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o 
/Library/Developer/CommandLineTools/usr/bin/cc -DAPPLE -I../../third_party/zlib -Ithird_party/zlib -I../../ -isystem ../../third_party/tinyxml2 -g -fPIC   -fno-exceptions -MD -MT third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o -MF third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o.d -o third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o   -c ../../third_party/zlib/adler32.c
error: unable to open output file 'third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o': 'Operation not permitted'
1 error generated.
[3/160] Building C object third_party/...ib/CMakeFiles/zlibstatic.dir/infback.o
FAILED: third_party/zlib/CMakeFiles/zlibstatic.dir/infback.o 
/Library/Developer/CommandLineTools/usr/bin/cc -DAPPLE -I../../third_party/zlib -Ithird_party/zlib -I../../ -isystem ../../third_party/tinyxml2 -g -fPIC   -fno-exceptions -MD -MT third_party/zlib/CMakeFiles/zlibstatic.dir/infback.o -MF third_party/zlib/CMakeFiles/zlibstatic.dir/infback.o.d -o third_party/zlib/CMakeFiles/zlibstatic.dir/infback.o   -c ../../third_party/zlib/infback.c
error: unable to open output file 'third_party/zlib/CMakeFiles/zlibstatic.dir/infback.o': 'Operation not permitted'
1 error generated.
[4/160] Building C object third_party/...ib/CMakeFiles/zlibstatic.dir/inffast.o
FAILED: third_party/zlib/CMakeFiles/zlibstatic.dir/inffast.o 
/Library/Developer/CommandLineTools/usr/bin/cc -DAPPLE -I../../third_party/zlib -Ithird_party/zlib -I../../ -isystem ../../third_party/tinyxml2 -g -fPIC   -fno-exceptions -MD -MT third_party/zlib/CMakeFiles/zlibstatic.dir/inffast.o -MF third_party/zlib/CMakeFiles/zlibstatic.dir/inffast.o.d -o third_party/zlib/CMakeFiles/zlibstatic.dir/inffast.o   -c ../../third_party/zlib/inffast.c
error: unable to open output file 'third_party/zlib/CMakeFiles/zlibstatic.dir/inffast.o': 'Operation not permitted'
1 error generated.
[5/160] Building C object third_party/...ib/CMakeFiles/zlibstatic.dir/inflate.o
[... cut ...]
/Library/Developer/CommandLineTools/usr/bin/cc -DAPPLE -I../../third_party/zlib -Ithird_party/zlib -I../../ -isystem ../../third_party/tinyxml2 -g -fPIC   -fno-exceptions -MD -MT third_party/zlib/CMakeFiles/zlibstatic.dir/inftrees.o -MF third_party/zlib/CMakeFiles/zlibstatic.dir/inftrees.o.d -o third_party/zlib/CMakeFiles/zlibstatic.dir/inftrees.o   -c ../../third_party/zlib/inftrees.c
error: unable to open output file 'third_party/zlib/CMakeFiles/zlibstatic.dir/inftrees.o': 'Operation not permitted'
1 error generated.
ninja: build stopped: subcommand failed.
make: *** [default] Error 1
```

ayaanhaider, 2019-01-16:
Also just to add, I wasn't able to download some dependencies on Mac since they weren't available for download using brew.

JonasVautherin, 2019-01-16:
> I removed the line you told me to but still no luck and got a similar error message.

What was that message? Another unknown flag?

> I got this error while running it on the second MacBook

You have permission errors there. Can you try to clean and build again?

```
make distclean
make
```

ayaanhaider, 2019-01-16:
> What was that message? Another unknown flag?

I got this after removing `-Wno-unused-lambda-capture`
```
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.dylib (found version "7.51.0") 
CMake Deprecation Warning at third_party/tinyxml2/CMakeLists.txt:11 (cmake_policy):
  The OLD behavior for policy CMP0063 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - failed
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Version: v0.12.0-2-gd842bd2b-dirty
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- Found PythonInterp: /Library/Frameworks/Python.framework/Versions/2.7/bin/python (found version "2.7.14") 
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   zlib

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/ayaanhaider/DronecodeSDK/build/default
Build dir: build/default
[36/160] Linking C shared library third_party/zlib/libz.1.2.11.dylib
FAILED: third_party/zlib/libz.1.2.11.dylib 
: && /Library/Developer/CommandLineTools/usr/bin/cc -g  -dynamiclib -Wl,-headerpad_max_install_names -compatibility_version 1.0.0 -current_version 1.2.11 -o third_party/zlib/libz.1.2.11.dylib -install_name /Users/ayaanhaider/DronecodeSDK/build/default/third_party/zlib/libz.1.dylib third_party/zlib/CMakeFiles/zlib.dir/adler32.o third_party/zlib/CMakeFiles/zlib.dir/compress.o third_party/zlib/CMakeFiles/zlib.dir/crc32.o third_party/zlib/CMakeFiles/zlib.dir/deflate.o third_party/zlib/CMakeFiles/zlib.dir/gzclose.o third_party/zlib/CMakeFiles/zlib.dir/gzlib.o third_party/zlib/CMakeFiles/zlib.dir/gzread.o third_party/zlib/CMakeFiles/zlib.dir/gzwrite.o third_party/zlib/CMakeFiles/zlib.dir/inflate.o third_party/zlib/CMakeFiles/zlib.dir/infback.o third_party/zlib/CMakeFiles/zlib.dir/inftrees.o third_party/zlib/CMakeFiles/zlib.dir/inffast.o third_party/zlib/CMakeFiles/zlib.dir/trees.o third_party/zlib/CMakeFiles/zlib.dir/uncompr.o third_party/zlib/CMakeFiles/zlib.dir/zutil.o -L/Users/ayaanhaider/DronecodeSDK/third_party/tinyxml2  && :
ld: can't write output file: third_party/zlib/libz.1.2.11.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[41/160] Building CXX object core/CMakeFiles/dronecode_sdk.dir/system.cpp.o
ninja: build stopped: subcommand failed.
make: *** [default] Error 1
```

ayaanhaider, 2019-01-16:
I tried the following and I got a successful build on my clang 9 MacBook:
```
make distclean
make
```

On my second laptop I tried cleaning but got an error message: 

```
- Generating done
-- Build files have been written to: /Users/Shayaan/DronecodeSDK/build/default
Build dir: build/default
[86/160] Building CXX object CMakeFile...runner.dir/core/thread_pool_test.cpp.o
FAILED: CMakeFiles/unit_tests_runner.dir/core/thread_pool_test.cpp.o
/Library/Developer/CommandLineTools/usr/bin/c++  -DAPPLE -DDRONECODE_SDK_VERSION=\"v0.12.0-dirty\" -DFAKE_TIME=1 -D_CRT_SECURE_NO_WARNINGS -I../../third_party/zlib -I../../core -I../../third_party/mavlink/include -I../../core/include -I../../plugins/mission/include -I../../plugins/mission/third_party/json11 -I../../plugins/camera/include -I../../plugins/calibration/include -isystem ../../third_party/tinyxml2 -isystem ../../third_party/gtest/googletest/include -isystem ../../third_party/gtest/googlemock/include -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wno-missing-braces -Wno-unused-parameter -Wno-old-style-cast -std=gnu++11 -MD -MT CMakeFiles/unit_tests_runner.dir/core/thread_pool_test.cpp.o -MF CMakeFiles/unit_tests_runner.dir/core/thread_pool_test.cpp.o.d -o CMakeFiles/unit_tests_runner.dir/core/thread_pool_test.cpp.o -c ../../core/thread_pool_test.cpp
../../core/thread_pool_test.cpp:67:17: error: lambda capture 'first' is not required to be captured for this use [-Werror,-Wunused-lambda-capture]
   tp.enqueue([first, &second]() { second += first; });
               ^
1 error generated.
[91/160] Building CXX object plugins/c...ecode_sdk_camera.dir/camera_impl.cpp.o
ninja: build stopped: subcommand failed.
make: *** [default] Error 1

```

Thanks so much for your help Jonas.

ayaanhaider, 2019-01-16:
Ah I see! Yes finally got it to work on the clang 10 as well 😄 Thanks a lot @JonasVautherin

closed 2019-01-16 by ayaanhaider

rajhlinux, 2022-04-20:
I was also getting this same error. I was able to suppress it from halting compilation using this flag:

-Wno-unused-lambda-capture