mavlink/MAVSDK #651: Expose temperature

sanderux, 2019-01-23:
I would like to be able to read HIGHRES_IMU.temperature

JonasVautherin, 2019-01-24:
@julianoes: is that `Info` or `Telemetry`? Or something else?

julianoes, 2019-01-24:
I'd say that's for telemetry.

sanderux, 2019-01-24:
I might be completely off in the idea behind the SDK, but why would you not expose all mavlink topics and allow getting updates? You can still create wrappers around standard things but you are not limiting devs to access all telemetry information.

Same goes for sending stuff, its good to have wrappers for everything the SDK understands, but a generic message factory such as dronekit/pymavlink would serve as a fall-back for anything unknown to the SDK or custom added to Mavlink.

julianoes, 2019-01-24:
Yes, a "backdoor" like that is something we probably should do. But that's orthogonal to this request.

julianoes, 2019-10-10:
This has been added to the C++ API:
https://github.com/mavlink/MAVSDK/blob/221bd80fd96f5524da2a14f4f5f46eb667e411e4/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L177
but it's missing in Python, so it needs to be added to the proto file here:
https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto

closed 2024-05-27 by julianoes