mavlink/MAVSDK #653: Segfault if port failed to open

bresch, 2019-01-28:
I have a piece of code using the SDK that binds to an UDP MAVLink port generated by the MAVLink router.
Unfortunately, the port where I wanted to connect is already used by an other application. Since the connection failed, sending an action to the drone resulted in a segfault.
Would it be possible to throw an error to prevent that segfault?

```
[10:20:34|Error] bind error: Address already in use (udp_connection.cpp:82)
Connection failed: Bind error
[...] // some code running and using const Action::Result ams_result = _action->...
Segmentation fault (core dumped)
```

Thanks!

julianoes, 2019-01-28:
Would be great if you could add a stacktrace. Just add `gdb --args` before the command that you're running and then copy `backtrace` once it crashed.

bresch, 2019-01-30:
@julianoes Here it is!
```
Thread 1 "xxx" received signal SIGSEGV, Segmentation fault.
0x76e3db56 in std::unique_ptr<dronecode_sdk::ActionImpl, std::default_delete<dronecode_sdk::ActionImpl> >::get (this=0x4)
    at /usr/include/c++/5/bits/unique_ptr.h:305
305     /usr/include/c++/5/bits/unique_ptr.h: No such file or directory.
```
```
(gdb) backtrace
#0  0x76e3db56 in std::unique_ptr<dronecode_sdk::ActionImpl, std::default_delete<dronecode_sdk::ActionImpl> >::get (this=0x4)
    at /usr/include/c++/5/bits/unique_ptr.h:305
#1  0x76e3da32 in std::unique_ptr<dronecode_sdk::ActionImpl, std::default_delete<dronecode_sdk::ActionImpl> >::operator-> (
    this=0x4) at /usr/include/c++/5/bits/unique_ptr.h:299
#2  0x76e3d34a in dronecode_sdk::Action::auto_maneuver_system (this=0x0) at /root/dronecode_sdk/plugins/action/action.cpp:55
#3  0x000181dc in SDK_Interface::ams() ()
#4  0x000178fc in loop(Autopilot_Interface&, SDK_Interface&) ()
#5  0x000176ce in top(int, char**) ()
#6  0x00017bca in main ()
```

Thanks.

julianoes, 2019-01-30:
`auto_maneuver_system` that's not a thing I recognize? :astonished:

julianoes, 2019-01-30:
You can avoid the problem by checking the return value of creating the connection, and exit early if that happens.

However, it still probably should not crash if no valid connection is available. I'm checking that now.

julianoes, 2019-01-30:
The problem was in the software using this which did not initialize `action` if the connection failed.

closed 2019-01-30 by julianoes