mavlink/MAVSDK #663: Mission: provide API to clear mission

julianoes, 2019-02-11:
We should probably offer a method that allows to wipe any missions stored on the vehicle. Right now, we can achieve this by uploading an empty mission item list but that's not really visible.

An alternative would be to document what it means to upload an empty list.

The underlying message to send would be: https://mavlink.io/en/messages/common.html#MISSION_CLEAR_ALL

JonasVautherin, 2019-02-11:
How many missions can be stored on the vehicle?

julianoes, 2019-02-11:
Two but only one at a time. So let's say you're flying `0` and uploading a mission, then it will be stored to `1`. Once fully uploaded, everything will switch over to `1`.

hamishwillee, 2019-02-11:
FYI clear mission is documented in https://mavlink.io/en/services/mission.html#clear_mission. In the update for partial missions I note what happens when you upload an empty mission https://github.com/mavlink/mavlink-devguide/pull/157/files#diff-17277822f661b48f9c95c6f034ab0469R166

julianoes, 2019-02-12:
Thanks @hamishwillee !

julianoes, 2019-10-17:
Thanks @douglaswsilva yes.