mavlink/MAVSDK #672: Command id 0

michprev, 2019-02-15:
Hello,
I noticed that after initialization of Telemetry object `shared_ptr<Telemetry> telemetry = make_shared<Telemetry>(system);` there are sent some [COMMAND_LONG commands](https://mavlink.io/en/messages/common.html#COMMAND_LONG) with target system, target component and component id all set to 0. However, there does not seem to be a command with id 0 in the [list](https://mavlink.io/en/messages/common.html#mavlink-commands-mavcmd). What does it mean / why is it sent?

I am using revision 8556877 (v0.14.1).

Thank you.

julianoes, 2019-02-15:
Ok, and what is the command id?

I expect some commands being sent to system id 0 and component id 0 which just means to everyone, although we might want to make that more specific, I agree.

michprev, 2019-02-15:
Command id is 0 - that is the strangest. I will try to use MavlinkPassthrough plugin to make sure that the command is being sent with id 0.

michprev, 2019-02-15:
MavlinkPassthrough does not report any COMMAND_LONG with id 0. Also Wireshark does not see any datagram that would look like it.
I think I can clearly say that issue is not in your SDK. Thank you for your support!

closed 2019-02-15 by michprev

julianoes, 2019-02-18:
Ok, thanks for the feedback.