mavlink/MAVSDK #685: calibration without STATUSTEXT commands

michprev, 2019-02-26:
While building my own flight stack I have hit an issue with calibration.

```C
void calibration_callback(const Calibration::Result result, const Calibration::ProgressData progress)
{
    cout << "progress " << progress.progress << " result " << (int)result << endl;
}

...

auto calibration = make_shared<Calibration>(system);
calibration->calibrate_accelerometer_async(calibration_callback);
```

The situation is:

1. my quad receives [COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG) with [MAV_CMD_PREFLIGHT_CALIBRATION](https://mavlink.io/en/messages/common.html#MAV_CMD_PREFLIGHT_CALIBRATION) id
2. it starts sending [COMMAND_ACK](https://mavlink.io/en/messages/common.html#COMMAND_ACK) with MAV_RESULT_IN_PROGRESS as result
3. In the end it sends one [COMMAND_ACK](https://mavlink.io/en/messages/common.html#COMMAND_ACK) with MAV_RESULT_ACCEPTED

I am not sending [STATUSTEXT](https://mavlink.io/en/messages/common.html#STATUSTEXT) commands

The issues are:

1. The callback is never called. I think that it is related to
https://github.com/Dronecode/DronecodeSDK/blob/e5d4e0e8510cd02bfbfe5d629e1a2e4ff49549c9/core/mavlink_commands.cpp#L180-L198
Commands with MAV_RESULT_IN_PROGRESS result do not get propagated - that it known issue. The problem is that `calibration_callback` is not called even when the last command with MAV_RESULT_ACCEPTED is received. That can be fixed by inserting something like
`call_user_callback(_calibration_callback, Calibration::Result ::SUCCESS, Calibration::ProgressData(false, NAN, false, ""));` here:
https://github.com/Dronecode/DronecodeSDK/blob/e5d4e0e8510cd02bfbfe5d629e1a2e4ff49549c9/plugins/calibration/calibration_impl.cpp#L203-L205

2. It is not possible to run the calibration for the second time simply because `_state` is still set to `State::ACCELEROMETER_CALIBRATION`. It should be cleared `_state = State::NONE;` here:
https://github.com/Dronecode/DronecodeSDK/blob/e5d4e0e8510cd02bfbfe5d629e1a2e4ff49549c9/plugins/calibration/calibration_impl.cpp#L203-L205

If the current implementation is working for you then it is probably because the flight stack also sends [STATUSTEXT](https://mavlink.io/en/messages/common.html#STATUSTEXT) commands.
https://github.com/Dronecode/DronecodeSDK/blob/e5d4e0e8510cd02bfbfe5d629e1a2e4ff49549c9/plugins/calibration/calibration_impl.cpp#L261-L338

julianoes, 2019-02-27:
Your own flight stack? :astonished: Is it  open source?

So your observation is correct, and the implementation using these statustexts is really clunky in PX4. We would like to come up with a protocol of MAVLink messages that make the communication of a calibration easier and clearer but still flexible for different sensors, calibration methods, airframes, etc.. That's a big todo but will probably take some time.

michprev, 2019-03-02:
It is currently closed course. I will consider open sourcing it once I feel comfortable with it :smile: There is not much implemented yet.

Unfortunately I don't own any flight controller that would work with MAVLink but I wonder what is need by calibration methods that STATUSTEXT can satisfy but COMMAND_ACK can not.

JonasVautherin, 2019-05-16:
Should we keep this opened here, or is it a MAVLink issue?

julianoes, 2019-05-16:
The events interface will eventually fix this. @bkueng is working on it.

closed 2019-05-16 by julianoes