mavlink/MAVSDK #689: Mission progress won't send event until aircraft reaches the following waypoint

chenny408, 2019-03-06:
This happens for mission items with loiterTimeS = .nan, cameraAction = stopPhotoInterval/no action. We do not get mission progress event until aircraft reaches the following waypoint.

We get a mission progress event every time the drone reaches a waypoint.
The issue seen is, if you have a waypoint mission (for example 10 waypoints), where 5th waypoint has 'loiterTimeS = .nan' & 'cameraAction = .none' and the 6th waypoint has 'loiterTimeS = 1' & 'cameraAction = .takePhoto', then when the vehicle reaches 5th waypoint, we do not receive mission progress event and when the vehicle reaches 6th waypoint, we receive the mission progress event twice.
This happens for mission items with loiterTimeS = .nan, cameraAction = stopPhotoInterval/no action.

JonasVautherin, 2019-03-07:
@julianoes: Would you mind having a look at that?

julianoes, 2019-03-07:
So the way it is supposed to work is that you only receive a mission progress for a mission item which is actually a waypoint, so a mission item with lat/lon defined.

Would it be possible to share a minimal list of mission items to reproduce this?

julianoes, 2020-11-20:
This is old, and has likely been fixed in the meantime, closing.

closed 2020-11-20 by julianoes