mavlink/MAVSDK #691: Inconsistent documentation in Action class.

jwidauer, 2019-03-07:
In the `Action` class the function `goto_location()`s documentation specifies:  
`Send command to reposition the vehicle to a specific WGS84 global position.`

But then underneath it it states:  
`float altitude_amsl_m - Altitude AMSL in meters`

Does this mean the `altitude_amsl_m` will be converted to WGS84 internally or is this a mistake in the documentation?

julianoes, 2019-03-07:
Good point, thanks for raising this.

I think this is the same that @mzahana found in Dronecode/DronecodeSDK#485 and lead me to create this issue:
https://github.com/PX4/Firmware/issues/10246

Given that, we should make sure that's documented correctly (for now) here.

julianoes, 2019-03-07:
I guess the global position would refer to latitude/longitude but is excluding the  altitude.

hamishwillee, 2019-03-07:
Almost certainly it is WGS84 position, and AMSL altitude, because that is the underlying definition of the MAVLink frames: https://mavlink.io/en/messages/common.html#MAV_FRAME

This is complicated further by the fact that the actual frame might not quite match even that, depending on the positioning hardware (anecdotally, came up in a MAVLink call).

PR #693 (2019-03-15): action: improve docs for goto altitude
Currently, PX4 does not let us specifiy absolute altitudes for goto, so AMSL (above mean sea level) is all we have.

Closes #691.
closed 2019-03-15 by julianoes