mavlink/MAVSDK #695: Camera range options not supported

julianoes, 2019-03-14:
We should support camera settings that are not an enum option but rather a range.

An example for such a param can be found in the camera manager:

https://github.com/Dronecode/camera-manager/blob/eef47667506eb97e6ee7349489b7738a825dae14/samples/def/camera-def-uvc.xml#L57-L59

julianoes, 2019-03-15:
@JonasVautherin we'll have to figure out the proto for this first, I think.

JonasVautherin, 2019-03-15:
Agreed. Is it already needed for a camera we support?

julianoes, 2019-03-15:
It's used a lot in camera-manager, so it's something needed for https://github.com/Auterion/mavlink-testing-suite which tests against camera-manager.

JonasVautherin, 2019-07-03:
Thinking about extending to other types (and ranges). My feeling is that we want to keep only one list of settings (`SubscribePossibleSettingOptions`), and therefore we want to keep `Setting` and `Option`.

What about modifying `Option` to something like:

```proto
message Option {
    string option_value = 1;
    string option_description = 2;
}
```

Without specifying the type on the client side (so the client sets a string). The parsing has to be done on the camera side, and can return a `WRONG_ARGUMENT`.

Then, `SettingOptions` has to allow for ranges:

```proto
message SettingOptions {
    string setting_id = 1;
    string setting_description = 2;
    repeated Option options = 3;
    bool is_range = 4;
}
```

If `is_range == true`, then `repeated Option options` must contain exactly 2 options with `option_value` set respectively to [min, max].

I believe we also want to specify a reasonable interval for a range, and possibly a type, right?

julianoes, 2019-07-03:
Hm, that's a bit tricky if you ask me but would be one way.

JonasVautherin, 2019-07-03:
What would be an easier way? Whatever else I can think of implies adding messages per type, and that ends up giving stuff like `SubscribeIntSettings`, `SubscribeEnumSettings`, ... We cannot do fancy `OneOf` or stuff like that, because in the end we have to be able to translate that to all the languages we support.

julianoes, 2019-07-03:
`SubscribeRangeSettings`? I see the problem, what do you do with the types.

JonasVautherin, 2019-07-04:
> SubscribeRangeSettings? I see the problem, what do you do with the types.

Exactly. In Java/Swift, we get a stream of one type. Having a supertype (`Object` or `Any`) is quite similar to having a `String`, in the end.

The camera knows that type it wants for each param, so it can check it and return an error for a wrong argument. Which is probably already the case for, say, a range: if I need an `int` betweeen 0 and 10 and I send 15, the camera will complain (`WRONG_ARGUMENT`?). I don't see a problem with doing that as well if I send "fifteen plz".

Same on the client side, when receiving a value. I don't see it as a big problem if users receive "8" (string) and need to cast it to 8 (int) themselves, given that the camera should never send an invalid value like "eight, I think" (string).

That's not the most beautiful interface ever, but if it contains the information needed to write a dynamic interface on the client side while being auto-generated on the SDK side, I believe it is not too bad.

I would be interested in @byuarus and @douglaswsilva opinions, as they have implemented a client side (not yet for the ranges, obviously).

julianoes, 2019-07-05:
I'm almost convinced :smile:.

One more thing: I think we also need to be able to specify the interval, so what about:

If `is_range == true`, then `repeated Option options` must contain 2 options with `option_value` set respectively to [min, max] or 3 options with `option_value` set respectively to [min, interval, max].

JonasVautherin, 2019-07-05:
That would make sense to me :+1:.

Possibly we should also provide the type of the values, in case the client needs that? Let's wait for @byuarus and @douglaswsilva for that, maybe :D

byuarus, 2019-07-15:
> Possibly we should also provide the type of the values, in case the client needs that? 

@JonasVautherin Yes, you're right. We need this :) 

> If is_range == true, then repeated Option options must contain 2 options with option_value set respectively to [min, max] or 3 options with option_value set respectively to [min, interval, max].

@julianoes  This will work for us.

Currently, for Sony R10C camera, we receive a json file with all possible settings. For R10C they provide the range as an enum, this will work for us too. I would say that it's preferred way of presenting range settings as it will make less likely that client-side will send some weird values. It will save from issues with value type conversions (String/Int/Float/Double) from the UI layer to SDK.

Example of JSON file:
[cameraSettingsJSON.txt](https://github.com/mavlink/MAVSDK/files/3394403/cameraSettingsJSON.txt)

PR #791 (2019-07-22): camera: support for range based params
This brings initial support in the camera plugin for range based parameters.

Range based parameters are parameters without a predefined option list like an enum but rather a min, max, step size and a type.

FYI @dogmaphobic 

Fixes #695.
closed 2019-07-22 by julianoes