mavlink/MAVSDK #698: Cancel mission upload and download lead to crashes

julianoes, 2019-03-15:
According to @douglaswsilva: 

> cancel mission upload and download, which are working but sometimes crash the app. I had to add a 5 second delay between canceling and uploading mission in order to prevent the crash, so it might be some async weird behavior.

Stacktrace:
```
#1	0x000000012729b07c in bool dronecode_sdk::Any::is<unsigned char>() const at /Users/douglas/Documents/DronecodeSDK/build/ios/../../core/any.h:27
#2	0x000000012729adbc in dronecode_sdk::MAVLinkParameters::ParamValue::is_same_type(dronecode_sdk::MAVLinkParameters::ParamValue const&) const at /Users/douglas/Documents/DronecodeSDK/build/ios/../../core/mavlink_parameters.h:356
#3	0x0000000127295430 in dronecode_sdk::MAVLinkParameters::ParamValue::operator==(dronecode_sdk::MAVLinkParameters::ParamValue const&) const at /Users/douglas/Documents/DronecodeSDK/build/ios/../../core/mavlink_parameters.h:377
#4	0x00000001272c3ed0 in dronecode_sdk::CameraDefinition::get_possible_options(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<dronecode_sdk::MAVLinkParameters::ParamValue, std::__1::allocator<dronecode_sdk::MAVLinkParameters::ParamValue> >&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_definition.cpp:506
#5	0x00000001272944c0 in dronecode_sdk::CameraImpl::get_possible_options(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<dronecode_sdk::Camera::Option, std::__1::allocator<dronecode_sdk::Camera::Option> >&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_impl.cpp:1129
#6	0x000000012729768c in dronecode_sdk::CameraImpl::notify_possible_setting_options() at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_impl.cpp:1327
#7	0x00000001272971b4 in dronecode_sdk::CameraImpl::subscribe_possible_setting_options(std::__1::function<void (std::__1::vector<dronecode_sdk::Camera::SettingOptions, std::__1::allocator<dronecode_sdk::Camera::SettingOptions> >)> const&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_impl.cpp:1267
#8	0x0000000127285ae4 in dronecode_sdk::Camera::subscribe_possible_setting_options(std::__1::function<void (std::__1::vector<dronecode_sdk::Camera::SettingOptions, std::__1::allocator<dronecode_sdk::Camera::SettingOptions> >)> const&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera.cpp:166
#9	0x0000000126fad99c in dronecode_sdk::backend::CameraServiceImpl<dronecode_sdk::Camera>::SubscribePossibleSettingOptions(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../backend/src/plugins/camera/camera_service_impl.h:628
#10	0x0000000127084648 in decltype(*(std::__1::forward<dronecode_sdk::rpc::camera::CameraService::Service*>(fp0)).*fp(std::__1::forward<grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(fp1))) std::__1::__invoke<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::* const&)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*), dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*, void>(grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::* const&&&)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*), dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/type_traits:4369
#11	0x0000000127084594 in std::__1::__invoke_return<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*), dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>::type std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>::operator()<dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) const [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/functional:1273
#12	0x0000000127084550 in decltype(std::__1::forward<std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&>(fp)(std::__1::forward<dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(fp0))) std::__1::__invoke<std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&, dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&&&, dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/type_traits:4428
#13	0x00000001270844fc in grpc::Status std::__1::__invoke_void_return_wrapper<grpc::Status>::__call<std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&, dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&&&, dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibl [... cut ...] ned] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/thread:342
#25	0x000000012734af04 in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (grpc::ThreadManager::WorkerThread::*)(), grpc::ThreadManager::WorkerThread*> >(void*) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/thread:352
```

julianoes, 2019-03-18:
It's weird because this stacktrace is for a camera crash, and not a mission cancel crash.

douglaswsilva, 2019-03-18:
My bad, let me send you the right one.

douglaswsilva, 2019-03-18:
`#26. Crashed: Thread
0  libsystem_pthread.dylib        0x1c4dbde9c pthread_mutex_lock$VARIANT$mp + 126
1  libc++.1.dylib                 0x1c4340b30 std::__1::mutex::lock() + 12
2  libc++.1.dylib                 0x1c430cbd8 std::__1::__assoc_sub_state::set_value() + 36
3  backend                        0x1116f90dc dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)::operator()(dronecode_sdk::Mission::Result) const + 88
4  backend                        0x1116f9078 void std::__1::__invoke_void_return_wrapper<void>::__call<dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)&, dronecode_sdk::Mission::Result>(dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)&&&, dronecode_sdk::Mission::Result&&) + 84
5  backend                        0x1116f8eb4 std::__1::__function::__func<dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result), std::__1::allocator<dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)>, void (dronecode_sdk::Mission::Result)>::operator()(dronecode_sdk::Mission::Result&&) + 60
6  backend                        0x111a2e8b0 std::__1::function<void (dronecode_sdk::Mission::Result)>::operator()(dronecode_sdk::Mission::Result) const + 76
7  backend                        0x111a2e858 dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0::operator()() const + 28
8  backend                        0x111a2e830 void std::__1::__invoke_void_return_wrapper<void>::__call<dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0&>(dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0&&&) + 48
9  backend                        0x111a2e4cc std::__1::__function::__func<dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0, std::__1::allocator<dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0>, void ()>::operator()() + 44
10 backend                        0x1119a06e4 std::__1::function<void ()>::operator()() const + 60
11 backend                        0x1119a0204 dronecode_sdk::ThreadPool::worker() + 236
12 backend                        0x1119a2344 void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> >(void*) + 472
13 libsystem_pthread.dylib        0x1c4dc425c _pthread_body + 128
14 libsystem_pthread.dylib        0x1c4dc41bc _pthread_start + 48
15 libsystem_pthread.dylib        0x1c4dc7cf4 thread_start + 4`

douglaswsilva, 2019-03-18:
`inline void UploadMissionResponse::set_allocated_mission_result(::dronecode_sdk::rpc::mission::MissionResult* mission_result) {
  ::google::protobuf::Arena* message_arena = GetArenaNoVirtual();
  if (message_arena == NULL) {
    delete mission_result_;
  }
  if (mission_result) {
    ::google::protobuf::Arena* submessage_arena = NULL;
    if (message_arena != submessage_arena) {
      mission_result = ::google::protobuf::internal::GetOwnedMessage(
          message_arena, mission_result, submessage_arena);
    }
    
  } else {
    
  }
  mission_result_ = mission_result;
  // @@protoc_insertion_point(field_set_allocated:dronecode_sdk.rpc.mission.UploadMissionResponse.mission_result)
}`

It crashes here, at `delete mission_result_;`

julianoes, 2019-03-19:
Great, thanks @douglaswsilva.

julianoes, 2019-03-19:
@JonasVautherin any idea what this could be?

JonasVautherin, 2019-03-28:
@julianoes: Can we close this following #701?

julianoes, 2019-03-28:
Let @douglaswsilva test it and confirm. Probably best to cut a release to test it.

douglaswsilva, 2019-03-28:
```
#0	0x000000018d023104 in __pthread_kill ()
#1	0x000000018d09f020 in pthread_kill$VARIANT$mp ()
#2	0x000000018cf7ad78 in abort ()
#3	0x000000018c644f78 in abort_message ()
#4	0x000000018c645108 in default_terminate_handler() ()
#5	0x000000018c65de68 in _objc_terminate() ()
#6	0x000000018c6510fc in std::__terminate(void (*)()) ()
#7	0x000000018c650a40 in __cxa_throw ()
#8	0x000000018c633690 in std::__1::__throw_system_error(int, char const*) ()
#9	0x000000018c624b48 in std::__1::mutex::lock() ()
#10	0x000000018c5f0bd8 in std::__1::__assoc_sub_state::set_value() ()
#11	0x0000000103b0700c in dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)::operator()(dronecode_sdk::Mission::Result) const ()
#12	0x0000000103b06fa8 in void std::__1::__invoke_void_return_wrapper<void>::__call<dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)&, dronecode_sdk::Mission::Result>(dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)&&&, dronecode_sdk::Mission::Result&&) ()
#13	0x0000000103b06de4 in std::__1::__function::__func<dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result), std::__1::allocator<dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::uploadMissionItems(std::__1::vector<std::__1::shared_ptr<dronecode_sdk::MissionItem>, std::__1::allocator<std::__1::shared_ptr<dronecode_sdk::MissionItem> > > const&, dronecode_sdk::rpc::mission::UploadMissionResponse*, std::__1::promise<void>&) const::'lambda'(dronecode_sdk::Mission::Result)>, void (dronecode_sdk::Mission::Result)>::operator()(dronecode_sdk::Mission::Result&&) ()
#14	0x0000000103fa5d58 in std::__1::function<void (dronecode_sdk::Mission::Result)>::operator()(dronecode_sdk::Mission::Result) const ()
#15	0x0000000103fa5d00 in dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0::operator()() const ()
#16	0x0000000103fa5cd8 in void std::__1::__invoke_void_return_wrapper<void>::__call<dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0&>(dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0&&&) ()
#17	0x0000000103fa5974 in std::__1::__function::__func<dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0, std::__1::allocator<dronecode_sdk::MissionImpl::report_mission_result(std::__1::function<void (dronecode_sdk::Mission::Result)> const&, dronecode_sdk::Mission::Result)::$_0>, void ()>::operator()() ()
#18	0x0000000103b2ad0c in std::__1::function<void ()>::operator()() const ()
#19	0x0000000103f08fbc in dronecode_sdk::ThreadPool::worker() ()
#20	0x0000000103f0b078 in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> >(void*) ()
#21	0x000000018d0a825c in _pthread_body ()
#22	0x000000018d0a81bc in _pthread_start ()
#23	0x000000018d0abcf4 in thread_start ()
```

douglaswsilva, 2019-03-28:
```
```

PR #714 (2019-04-02): mission: fix double call of result callback
We need to make sure we set the callback to nullptr while we have the lock. This way we make sure we only ever call the callback once and don't upset the API user.

Fixes #698.
closed 2019-04-02 by julianoes