mavlink/MAVSDK #703: Automate dependencies for Linux and Windows

hamishwillee, 2019-03-21:
When building the MAVLink test suite setup fails on Windows and Linux because there are missing dependencies. As discussed  [here](https://github.com/Auterion/mavlink-testing-suite/issues/13#issuecomment-470413491) it would be better if the SDK is able to install its own dependencies.

@julianoes Just a reminder :-)

JonasVautherin, 2019-03-21:
I understand the issue for the MAVLink test suite, but I kind of disagree on the solution here. I would say that the MAVLink test suite should `find_package(DronecodeSDK REQUIRED)` instead of building it itself.

This said, I'm working on moving the dependencies out of the SDK CMakeLists and having some kind of helper to build them if necessary (but relying on `find_package` everywhere). And at the same time my goal is to go towards depending on the SDK with `find_package(DronecodeSDK REQUIRED)`, which would solve this issue.

hamishwillee, 2019-03-25:
@JonasVautherin I don't mind where or how the issue is solved, but it has to be solved in either the test suite or here :-). Work it out with @julianoes and let me know :-)
In the short term, this could just be addition to the readme in the test suite.

PR #724 (2019-05-30): Reorganize deps
This is moving the dependencies out of the project and adding them using `find_package` instead. Because we still want to provide a way to compile the dependencies from sources, it is organized as follows:

```
- superbuild
    |
    --- third_party
    |
    --- src
```

The superbuild adds both. Note that `third_party` builds the dependencies at configure time. 

To be done before merging this PR:
- [x] Fix all CI scripts
- [x] Add build for iOS
- [x] Update documentation (by adding a Makefile showing how to use CMake)
- [x] Test iOS build with Swift SDK
- [ ] ~Add build for  [... cut]
closed 2019-05-30 by julianoes