mavlink/MAVSDK #708: How to cross-compile dronecodesdk to run on armv9?

addriumruss, 2019-03-26:
Anyone could help? 
Thanks very much!

addriumruss, 2019-03-26:
I used an arm-none-linux-gnueabi-g++, it doesn't support c++11. 
environment: centos6.9 ,  dronecodesdk 2018.12
this is init.sh: 
#/bin/sh 
export CC=/home/russ/arm/arm-2009q1/bin/arm-none-linux-gnueabi-gcc
export CXX=/home/russ/arm/arm-2009q1/bin/arm-none-linux-gnueabi-g++
cmake ..
# make

and this is the cmake result:
[root@localhost build-arm]# ./init.sh | tee error
-- The C compiler identification is GNU 4.3.3
-- The CXX compiler identification is GNU 4.3.3
-- Check for working C compiler: /home/russ/arm/arm-2009q1/bin/arm-none-linux-gnueabi-gcc
-- Check for working C compiler: /home/russ/arm/arm-2009q1/bin/arm-none-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/russ/arm/arm-2009q1/bin/arm-none-linux-gnueabi-g++
-- Check for working CXX compiler: /home/russ/arm/arm-2009q1/bin/arm-none-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.so (found version "7.19.7") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - done
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
CMake Error in plugins/mission/third_party/json11/CMakeLists.txt:
  Target "json11" requires the language dialect "CXX11" (with compiler
  extensions), but CMake does not know the compile flags to use to enable it.

-- Generating done
-- Build files have been written to: /home/russ/px4/DronecodeSDK/build-arm

addriumruss, 2019-03-26:
anyone can help ?

shrit, 2019-03-26:
I think your compiler needs to be compatible with c++11, which mean at least version 4.8.1

addriumruss, 2019-03-26:
it's already 4.8.2：
[root@localhost temp]# g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/opt/rh/devtoolset-2/root/usr/libexec/gcc/i686-redhat-linux/4.8.2/lto-wrapper
Target: i686-redhat-linux
Configured with: ../configure --prefix=/opt/rh/devtoolset-2/root/usr --mandir=/opt/rh/devtoolset-2/root/usr/share/man --infodir=/opt/rh/devtoolset-2/root/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --enable-languages=c,c++,fortran,lto --enable-plugin --with-linker-hash-style=gnu --enable-initfini-array --disable-libgcj --with-isl=/dev/shm/home/centos/rpm/BUILD/gcc-4.8.2-20140120/obj-i686-redhat-linux/isl-install --with-cloog=/dev/shm/home/centos/rpm/BUILD/gcc-4.8.2-20140120/obj-i686-redhat-linux/cloog-install --with-mpc=/dev/shm/home/centos/rpm/BUILD/gcc-4.8.2-20140120/obj-i686-redhat-linux/mpc-install --with-tune=generic --with-arch=i686 --build=i686-redhat-linux
Thread model: posix
gcc version 4.8.2 20140120 (Red Hat 4.8.2-15) (GCC)

JonasVautherin, 2019-03-26:
Seems like it is failing on json11. Did you try to cross-compile [`json11`](https://github.com/dropbox/json11) only, as a first step?

addriumruss, 2019-03-27:
@JonasVautherin Good idea! I try it now.

addriumruss, 2019-03-27:
I got this:
cc1plus: error: unrecognized command line option "-std=c++11"
cc1plus: error: unrecognized command line option "-std=c++11"
maybe I should change a cross-compile tool.

julianoes, 2019-03-27:
This looks like C++11 is not supported. Any chance you can use a more recent compiler?

addriumruss, 2019-03-27:
@julianoes Thanks!
I'm looking for arm-linux-gcc-5.4, found a x64 version, but the x86 version is needed. 
Still searching.

JonasVautherin, 2019-03-27:
BTW, do you know [crosstool-ng](https://crosstool-ng.github.io/docs/) and/or [dockcross](https://github.com/dockcross/dockcross)? Dockcross doesn't seem to have armv9 (yet), but it may be worth asking them what it would require.

addriumruss, 2019-03-28:
@JonasVautherin Thanks for crosstool-ng tip!
I got the arm-hisiv400-linux-gcc tool yestoday from the device manufacturer, and now blocked at curl.
this is the cmake command:
#/bin/sh 
export CC=arm-hisiv400-linux-gcc
export CXX=arm-hisiv400-linux-g++
cmake ..

this is the cmake result:
[root@localhost build-arm]# ./init.sh 
-- The C compiler identification is GNU 4.8.3
-- The CXX compiler identification is GNU 4.8.3
-- Check for working C compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-gcc
-- Check for working C compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-g++
-- Check for working CXX compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.so (found version "7.19.7") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - done
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/russ/px4/DronecodeSDK/build-arm
[root@localhost build-arm]#

and this is the make result:
[root@localhost build-arm]# make
Scanning dependencies of target zlibstatic
[  1%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o
[  2%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/compress.o
[  3%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/crc32.o
[  4%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/deflate.o
[  5%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/gzclose.o
[  6%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/gzlib.o
[  7%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/gzread.o
[  8%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/gzwrite.o
[  9%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/inflate.o
[ 10%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/infback.o
[ 11%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/inftrees.o
[ 12%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/inffast.o
[ 13%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/trees.o
[... 47 lines cut ...]
                 from /home/russ/px4/DronecodeSDK/core/curl_wrapper.cpp:3:
/home/russ/px4/DronecodeSDK/core/curl_include.h:7:23: fatal error: curl/curl.h: 没有那个文件或目录
 #include <curl/curl.h>
                       ^
compilation terminated.
make[2]: *** [core/CMakeFiles/dronecore.dir/curl_wrapper.cpp.o] 错误 1
make[1]: *** [core/CMakeFiles/dronecore.dir/all] 错误 2
make: *** [all] 错误 2

I'm sure that curl exist:
[root@localhost build-arm]# ll /usr/include/curl/
总用量 168
-rw-r--r--. 1 root root  7148 4月   3 2017 curlbuild-32.h
-rw-r--r--. 1 root root   164 4月   3 2017 curlbuild.h
-rw-r--r--. 1 root root 73086 4月   3 2017 curl.h
-rw-r--r--. 1 root root  8419 10月 29 2009 curlrules.h
-rw-r--r--. 1 root root  2799 11月  4 2009 curlver.h
-rw-r--r--. 1 root root  3526 5月  13 2008 easy.h
-rw-r--r--. 1 root root  2850 5月  20 2008 mprintf.h
-rw-r--r--. 1 root root 13039 5月  20 2008 multi.h
-rw-r--r--. 1 root root  1384 5月  20 2009 stdcheaders.h
-rw-r--r--. 1 root root 33416 1月  26 2009 typecheck-gcc.h
-rw-r--r--. 1 root root    15 12月 25 2007 types.h

Is it possible to disable the curl??

addriumruss, 2019-03-28:
Hi, 
I rebuild curl with arm gcc, and mark it in dronecodesdk/CMakeLists.txt:
"
cmake_minimum_required(VERSION 3.1)
include_directories("/home/russ/px4/curl7/include")
link_directories("/home/russ/px4/curl7/lib")
set(CMAKE_CXX_STANDARD 11)
.......
"
then compile it , still got error: 

[root@localhost build-arm]# more init.sh 
#/bin/sh 
export CC=arm-hisiv400-linux-gcc
export CXX=arm-hisiv400-linux-g++
cmake -D CMAKE_INSTALL_PREFIX=/home/russ/px4/sdk-deploy ..

[root@localhost build-arm]# ./init.sh 
-- The C compiler identification is GNU 4.8.3
-- The CXX compiler identification is GNU 4.8.3
-- Check for working C compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-gcc
-- Check for working C compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-g++
-- Check for working CXX compiler: /opt/hisi-linux/x86-arm/arm-hisiv400-linux/target/bin/arm-hisiv400-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CURL: /usr/lib/libcurl.so (found version "7.19.7") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - done
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Read version: 0.10.1
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test _flag_supported
-- Performing Test _flag_supported - Failed
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/russ/px4/DronecodeSDK/build-arm

[root@localhost build-arm]# make
Scanning dependencies of target zlibstatic
[  1%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/adler32.o
[  2%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/compress.o
[  3%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/crc32.o
[  3%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/deflate.o
[  4%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/gzclose.o
[  5%] Building C object third_party/zlib/CMakeFiles/zlibstatic.dir/gzlib.o
[... 91 lines cut ...]
     class promise;
           ^
/home/russ/px4/DronecodeSDK/core/system_impl.cpp: In member function ‘std::pair<dronecode_sdk::MAVLinkParameters::Result, float> dronecode_sdk::SystemImpl::get_param_float(const string&)’:
/home/russ/px4/DronecodeSDK/core/system_impl.cpp:657:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/russ/px4/DronecodeSDK/core/system_impl.cpp: In member function ‘std::pair<dronecode_sdk::MAVLinkParameters::Result, int> dronecode_sdk::SystemImpl::get_param_int(const string&)’:
/home/russ/px4/DronecodeSDK/core/system_impl.cpp:679:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/russ/px4/DronecodeSDK/core/system_impl.cpp: In member function ‘std::pair<dronecode_sdk::MAVLinkParameters::Result, float> dronecode_sdk::SystemImpl::get_param_ext_float(const string&)’:
/home/russ/px4/DronecodeSDK/core/system_impl.cpp:702:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/russ/px4/DronecodeSDK/core/system_impl.cpp: In member function ‘std::pair<dronecode_sdk::MAVLinkParameters::Result, int> dronecode_sdk::SystemImpl::get_param_ext_int(const string&)’:
/home/russ/px4/DronecodeSDK/core/system_impl.cpp:725:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1plus: all warnings being treated as errors
make[2]: *** [core/CMakeFiles/dronecode_sdk.dir/system_impl.cpp.o] 错误 1
make[1]: *** [core/CMakeFiles/dronecode_sdk.dir/all] 错误 2
make: *** [all] 错误 2

Now I'm really confused by these errors. 
Can anyone help figure out a way?  Any guide would be appreciated!

julianoes, 2019-03-28:
> Is it possible to disable the curl??

If you don't need the camera plugin (or at least not the camera definition functionality), you could for now remove the curl dependency. You'd  just have to go through the code yourself and add #ifdefs wherever needed.

addriumruss, 2019-03-28:
@julianoes Thank you so much, I'll try add #ifdefs to hide curl. 
The previous issue above seems caused by class MAVLinkParameters and class SystemImpl both referenced each other. And the code looks no wrong. 
Do you have any ideas about this?

julianoes, 2019-03-28:
That's odd. I'll need to try to compile with gcc 4.8.3.

addriumruss, 2019-03-28:
Thanks so much!

JonasVautherin, 2019-03-28:
I don't see errors other than `-Werror`. Maybe it's faster to remove `-Werror` for this build rather than adding `#ifdefs` everywhere.

julianoes, 2019-03-28:
@JonasVautherin this is a real error, no?
```
/home/russ/px4/DronecodeSDK/core/system_impl.cpp: In member function ‘std::pair<dronecode_sdk::MAVLinkParameters::Result, float> dronecode_sdk::SystemImpl::get_param_float(const string&)’:
/home/russ/px4/DronecodeSDK/core/system_impl.cpp:639:75: error: invalid use of incomplete type ‘class std::promise<std::pair<dronecode_sdk::MAVLinkParameters::Result, float> >’
auto prom = std::promise<std::pair<MAVLinkParameters::Result, float>>();
```

JonasVautherin, 2019-03-28:
Oh, I missed it somehow. My bad :-)

addriumruss, 2019-03-29:
Thanks! 
The environment is Centos6.9, 32bit vmware, with cross-compile  arm-hisiv400-linux-gcc 4.8.3.

addriumruss, 2019-04-10:
I think this issue is more related to the cross-compile tool, which is provided by the HiSi manufacturer, rather than the dronecode-sdk. 
I'll close the issue and try the cross-compile tool again.
Thanks very much to everyone~

closed 2019-04-10 by addriumruss