mavlink/MAVSDK #712: Camera crash in possibleSettings

douglaswsilva, 2019-03-28:
#1        0x000000012729b07c in bool dronecode_sdk::Any::is<unsigned char>() const at /Users/douglas/Documents/DronecodeSDK/build/ios/../../core/any.h:27
#2        0x000000012729adbc in dronecode_sdk::MAVLinkParameters::ParamValue::is_same_type(dronecode_sdk::MAVLinkParameters::ParamValue const&) const at /Users/douglas/Documents/DronecodeSDK/build/ios/../../core/mavlink_parameters.h:356
#3        0x0000000127295430 in dronecode_sdk::MAVLinkParameters::ParamValue::operator==(dronecode_sdk::MAVLinkParameters::ParamValue const&) const at /Users/douglas/Documents/DronecodeSDK/build/ios/../../core/mavlink_parameters.h:377
#4        0x00000001272c3ed0 in dronecode_sdk::CameraDefinition::get_possible_options(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<dronecode_sdk::MAVLinkParameters::ParamValue, std::__1::allocator<dronecode_sdk::MAVLinkParameters::ParamValue> >&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_definition.cpp:506
#5        0x00000001272944c0 in dronecode_sdk::CameraImpl::get_possible_options(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<dronecode_sdk::Camera::Option, std::__1::allocator<dronecode_sdk::Camera::Option> >&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_impl.cpp:1129
#6        0x000000012729768c in dronecode_sdk::CameraImpl::notify_possible_setting_options() at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_impl.cpp:1327
#7        0x00000001272971b4 in dronecode_sdk::CameraImpl::subscribe_possible_setting_options(std::__1::function<void (std::__1::vector<dronecode_sdk::Camera::SettingOptions, std::__1::allocator<dronecode_sdk::Camera::SettingOptions> >)> const&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera_impl.cpp:1267
#8        0x0000000127285ae4 in dronecode_sdk::Camera::subscribe_possible_setting_options(std::__1::function<void (std::__1::vector<dronecode_sdk::Camera::SettingOptions, std::__1::allocator<dronecode_sdk::Camera::SettingOptions> >)> const&) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../plugins/camera/camera.cpp:166
#9        0x0000000126fad99c in dronecode_sdk::backend::CameraServiceImpl<dronecode_sdk::Camera>::SubscribePossibleSettingOptions(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*) at /Users/douglas/Documents/DronecodeSDK/build/ios/../../backend/src/plugins/camera/camera_service_impl.h:628
#10        0x0000000127084648 in decltype(*(std::__1::forward<dronecode_sdk::rpc::camera::CameraService::Service*>(fp0)).*fp(std::__1::forward<grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(fp1))) std::__1::__invoke<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::* const&)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*), dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*, void>(grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::* const&&&)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*), dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/type_traits:4369
#11        0x0000000127084594 in std::__1::__invoke_return<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*), dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>::type std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>::operator()<dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) const [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/functional:1273
#12        0x0000000127084550 in decltype(std::__1::forward<std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&>(fp)(std::__1::forward<dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(fp0))) std::__1::__invoke<std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&, dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&&&, dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/type_traits:4428
#13        0x00000001270844fc in grpc::Status std::__1::__invoke_void_return_wrapper<grpc::Status>::__call<std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&, dronecode_sdk::rpc::camera::CameraService::Service*, grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*>(std::__1::__mem_fn<grpc::Status (dronecode_sdk::rpc::camera::CameraService::Service::*)(grpc::ServerContext*, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*)>&&&, dronecode_sdk::rpc::camera::CameraService::Service*&&, grpc::ServerContext*&&, dronecode_sdk::rpc::camera::SubscribePossibleSettingOptionsRequest const*&&, grpc::ServerWriter<dronecode_sdk::rpc::camera::PossibleSettingOptionsResponse>*&&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault [... cut ...] ] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/thread:342
#25        0x000000012734af04 in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (grpc::ThreadManager::WorkerThread::*)(), grpc::ThreadManager::WorkerThread*> >(void*) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/thread:352

douglaswsilva, 2019-03-28:
It happens sporadically, can't find a perfect reproducible way yet, but saw it happening sometimes upon subscription after app crash.

douglaswsilva, 2019-05-08:
@julianoes and @JonasVautherin Can you guys take a look at this when you have a chance. We've been getting it pretty often. Here are the logs we have:

[com.3dr.site-scan_issue_crash_2b4f63d856f348bc99285854652edd9d_DNE_0_v2.txt](https://github.com/Dronecode/DronecodeSDK/files/3159613/com.3dr.site-scan_issue_crash_2b4f63d856f348bc99285854652edd9d_DNE_0_v2.txt)
[com.3dr.site-scan_issue_crash_b7426f7b457f40e39884e53b6164dfa9_DNE_0_v2.txt](https://github.com/Dronecode/DronecodeSDK/files/3159614/com.3dr.site-scan_issue_crash_b7426f7b457f40e39884e53b6164dfa9_DNE_0_v2.txt)

julianoes, 2019-05-09:
@douglaswsilva thanks for the bump. I will look into this today. Maybe I can blame the camera firmware update, however, the SDK should still not crash because of it!

julianoes, 2019-05-10:
@douglaswsilva I have pushed a potential fix #746. It would be nice if you could test it and see if it fixes the crash.

douglaswsilva, 2019-05-10:
Will do next week. Thanks Julian!

closed 2019-05-14 by julianoes

JonasVautherin, 2019-05-14:
@douglaswsilva: Please reopen here if #746 doesn't solve it.

douglaswsilva, 2019-05-17:
Thanks @JonasVautherin . Sorry, I still haven't had time to get to this. Is it included in Eddie?

JonasVautherin, 2019-05-18:
I don't think it is in 0.16.0. I can try to make a new release soon (like next week).

PR #746 (2019-05-14): Fix camera thread-safety
This should fix potential thread safety issues in `CameraDefinition`. It might fix #712.